PJLab-ADG/SensorsCalibration #4: run_lidar2lidar failed

chenshiruchao, 2022-06-08:
Hello, I want to run ./bin/run_lidar2lidar in the /share/lidar2lidar/manual_calib, but it failed and prompt "./bin/run_lidar2lidar: error while loading shared libraries: /usr/lib/x86_64-linux-gnu/libGL.so.1: file too short".

Spring-lovely, 2022-06-09:
Hi,
I have updated the docker image v1 version, please use the command `docker pull scllovewkf/ opencalib :v1` to pull .

> Hello, I want to run ./bin/run_lidar2lidar in the /share/lidar2lidar/manual_calib, but it failed and prompt "./bin/run_lidar2lidar: error while loading shared libraries: /usr/lib/x86_64-linux-gnu/libGL.so.1: file too short".

Spring-lovely, 2022-06-09:
Hi,
I have updated the docker image v1 version, please use the command `docker pull scllovewkf/ opencalib:v1` to pull .
Thank you!

chenshiruchao, 2022-06-09:
thanks for your answer, After updating, it prompt "
left_lidar-to-top_center_lidar-extrinsic
lidar to lidar extrinsic:
  0.504271  -0.718711  -0.457892   -1.87468
  0.604509    0.67629  -0.394462   -1.86523
  0.598991 -0.0777196   0.783179  -0.527022
         0          0          0          1
Loading data completed!
=>Calibration scale Init!
terminate called after throwing an instance of 'std::runtime_error'
  what():  Pangolin X11: Failed to open X display
Aborted"

BayRanger, 2022-06-11:
> 

Hi,
To address this issue, I forked this  repo and add a script for docker visualization.
You could check: [https://github.com/BayRanger/SensorsCalibration](url)
And the usage has been added to README

Bests,
BR

chenshiruchao, 2022-06-13:
Thank you very much, it works for me.

closed 2022-06-14 by chenshiruchao

Gigalomanicx, 2025-03-09:
> > 
> 
> Hi, To address this issue, I forked this repo and add a script for docker visualization. You could check: [https://github.com/BayRanger/SensorsCalibration](url) And the usage has been added to README
> 
> Bests, BR

Thank you. The problem has been solved. It seems that the key issue is not the script difference, but that the container must be created through ./run_docker.sh（official repository.）. Then it can be run on a local machine with a graphical interface.