PJLab-ADG/SensorsCalibration #5: How to get the initial value of extrinsic parameters between lidar and camera

lidepin, 2022-06-09:
How can I get the initial value of extrinsic parameters based on my PNG and PCD between lidar and camera in lidar2camera project? So I think this is a tool for fine-tuning, not for calibration，right？

xiaokn, 2022-06-13:
> How can I get the initial value of extrinsic parameters based on my PNG and PCD between lidar and camera in lidar2camera project? So I think this is a tool for fine-tuning, not for calibration，right？

An initial value can be set according to the corresponding relationship of the sensor's coordinate system.

lidepin, 2022-06-13:
> 
可以举个例子吗，或者推荐一下相关计算初值的资料

lidepin, 2022-06-13:
> An initial value can be set according to the corresponding relationship of the sensor's coordinate system.

可以举个例子吗，或者推荐一下计算初值的相关资料，谢谢

zpengch, 2022-06-27:
> > > 如何在lidar2camera项目中根据我的激光雷达和相机之间的PNG和PCD获取外部参数的初始值？所以我认为这是一个微调的工具，而不是校准的工具，对吧？
> > 
> > 
> > 可以根据传感器坐标系的对应关系设置一个初始值。
> 
> 可以举个例子吗，或者推荐一下计算初值的相关资料，谢谢

就是根据你传感器的安装位置，给一个大概的初值，比如你认为传感器之间的旋转角为0度，就给0度，像激光雷达和相机的坐标系不一致，所以你的初值肯定要加一个旋转角

Binbin418916, 2022-07-22:
> An initial value can be set according to the corresponding relationship of the sensor's coordinate system.

也注意到了JointCalib外参初值的问题，请问初值的误差能容忍多少？如果没法量没有大概初值怎么办？

daikunpeng, 2022-08-21:
我注意到在摄像头内参标定中，标定后的distortion参数有5个，是[k1 k2 p1 p2 k3]吗？然而在 lidar2camera/manual_calib 中的初始值设定中，摄像头distortion初始值的设定只有四个值，他们的物理含义是哪几个？希望能够得到回复，谢谢。

ajiang17, 2022-08-29:
> 我注意到在摄像头内参标定中，标定后的distortion参数有5个，是[k1 k2 p1 p2 k3]吗？然而在 lidar2camera/manual_calib 中的初始值设定中，摄像头distortion初始值的设定只有四个值，他们的物理含义是哪几个？希望能够得到回复，谢谢。

同问， 如果有5个畸变包括径向畸变函数，请问如何设置json 配置文件？

callzhang, 2022-11-22:
> > 我注意到在摄像头内参标定中，标定后的distortion参数有5个，是[k1 k2 p1 p2 k3]吗？然而在 lidar2camera/manual_calib 中的初始值设定中，摄像头distortion初始值的设定只有四个值，他们的物理含义是哪几个？希望能够得到回复，谢谢。
> 
> 同问， 如果有5个畸变包括径向畸变函数，请问如何设置json 配置文件？

同问参数配置的文档，现在只是有个“data”的参数

ChenZhiKai66, 2023-04-17:
> An initial value can be set according to the corresponding relationship of the sensor's coordinate system.

Hello, may I ask:
I set the initial value to 0, Residual and Jacobian evaluation failed
When I set it to a non-zero value, tvec still outputs according to its initial value and is not optimized

chrisyan, 2024-11-29:
> > > > 如何在lidar2camera项目中根据我的激光雷达和相机之间的PNG和PCD获取外部参数的初始值？所以我认为这是一个微调的工具，而不是校准的工具，对吧？
> > > 
> > > 
> > > 可以根据传感器坐标系的对应关系设置一个初始值。
> > 
> > 
> > 可以举个例子吗，或者推荐一下计算初值的相关资料，谢谢
> 
> 就是根据你传感器的安装位置，给一个大概的初值，比如你认为传感器之间的旋转角为0度，就给0度，像激光雷达和相机的坐标系不一致，所以你的初值肯定要加一个旋转角

hello, 对于初值， 感觉很难用肉眼估计出相机和LiDAR的相对旋转和相对位置，不知道是否有更好的办法？

Somebo1, 2025-01-07:
> > > > > 如何在lidar2camera项目中根据我的激光雷达和相机之间的PNG和PCD获取外部参数的初始值？所以我认为这是一个微调的工具，而不是校准的工具，对吧？
> > > > 
> > > > 
> > > > 可以根据传感器坐标系的对应关系设置一个初始值。
> > > 
> > > 
> > > 可以举个例子吗，或者推荐一下计算初值的相关资料，谢谢
> > 
> > 
> > 就是根据你传感器的安装位置，给一个大概的初值，比如你认为传感器之间的旋转角为0度，就给0度，像激光雷达和相机的坐标系不一致，所以你的初值肯定要加一个旋转角
> 
> hello, 对于初值， 感觉很难用肉眼估计出相机和LiDAR的相对旋转和相对位置，不知道是否有更好的办法？

如果有条件，可以利用三维软件(如solidworks)对传感器平台建模。这样计算初值会很精确。