PJLab-ADG/SensorsCalibration #6: Somequestion

lidepin, 2022-06-09:
1、Why can't all interfaces be adjusted by keyboard input
2、Selecting four points in the radar2amera project and pressing Enter does not working

wqiancheng, 2022-06-10:
我也有这个问题。标完四个点之后，怎么调出面板呀。

ycdhqzhiai, 2022-06-10:
> 我也有这个问题。标完四个点之后，怎么调出面板呀。
https://github.com/PJLab-ADG/SensorsCalibration/blob/4fa5a9099b7f282491e6fe4f06163da6c8d5120f/radar2camera/manual_calib/include/birdview.hpp#L58
根据自己键盘修改值，或者自己设定

wqiancheng, 2022-06-11:
> > 我也有这个问题。标完四个点之后，怎么调出面板呀。
> > https://github.com/PJLab-ADG/SensorsCalibration/blob/4fa5a9099b7f282491e6fe4f06163da6c8d5120f/radar2camera/manual_calib/include/birdview.hpp#L58
> > 
> > 根据自己键盘修改值，或者自己设定

非常感谢，我打印了一下key的值，enter键是值是10.  现在可以用了

zengdangci, 2023-03-22:
[image]
为什么我修改key值后，按enter还是没有调出面板？

shophelper, 2023-06-28:
> 

您好，请问您的homography matrix是怎么得到的呀