PJLab-ADG/SensorsCalibration #7: How to display GUI

dangerous-xu, 2022-06-09:
I have linked the code to the container and compile the code. If I run the script, how to display the GUI manually adjust parameters.
thanks!

BayRanger, 2022-06-11:
Hi,
To address this issue, I forked this  repo and add a script for docker visualization.
You could check: [https://github.com/BayRanger/SensorsCalibration](url)
And the usage has been added to README

Bests,
BR

GuoLiuFang, 2022-06-11:
> 
one mistake; and still not work
[image]

[image]

BayRanger, 2022-06-18:
you could try using sudo before the command in the container.

sywin, 2022-07-07:
> Hi, To address this issue, I forked this repo and add a script for docker visualization. You could check: [https://github.com/BayRanger/SensorsCalibration](url) And the usage has been added to README
> 
> Bests, BR

Thanks! It's ok!

JiayinDeng, 2022-07-13:
Hello, I have the same issue and have tried almost all the methods I could find. 

Finally, I noticed that it may be a better choice to install the required dependencies on the host machine rather than on docker. So, I installed Pangolin on my host machine using source code and solved this issue.

AndrewJSong, 2022-07-13:
> Hello, I have the same issue and have tried almost all the methods I could find. However, none of them can work.
> 
> Finally, I noticed that it may be a better choice to install the required dependencies on the host machine rather than on docker. So, I installed Pangolin on my host machine using source code and solved this issue.

没错，使用docker尝试过在远程服务器/本机/虚拟机/有无显卡的机器上测试，均无法实现图形界面显示。最后还是自己安装PCL VTK OPENCV实现可视化界面

kapaper, 2022-08-15:
> 没错，使用docker尝试过在远程服务器/本机/虚拟机/有无显卡的机器上测试，均无法实现图形界面显示。最后还是自己安装PCL VTK OPENCV实现可视化界面

I also want to visualize the GUI in docker, can you offer some instructions or docker image?

NovemberChopin, 2022-08-24:
> 

你好，我在本地装了opencv2.4.13，pcl1.9.1但是lidar2camera和radar2camera的手动标定程序总是出错。你本地是怎么安装的啊

NovemberChopin, 2022-08-24:
> 没错，使用docker尝试过在远程服务器/本机/虚拟机/有无显卡的机器上测试，均无法实现图形界面显示。最后还是自己安装PCL VTK OPENCV实现可视化界面

你好，我在本地装了opencv2.4.13，pcl1.9.1但是lidar2camera和radar2camera的手动标定程序总是出错。你本地是怎么安装的啊

Lin239522, 2022-08-26:
> 你好，我也遇到了同样的问题，几乎所有的人都能找到解决办法，但是不能工作。，他们都。
> 
> 最后，我注意到在主机上而不是在 docker 上安装所需的选择依赖项可能是更好的。所以，我使用源代码在我的主机上安装了 Pangolin 并解决了这个问题。

What version of pangolin is it? My pangolin installation does not support C++11

AndrewJSong, 2022-09-01:
Ubuntu22 本地安装环境的相关依赖版本：
建议各位在虚拟机中尝试通过后再在实体机中安装，其中会有几个错误需要修改，百度很好解决，古德拉克
# pangolin 0.6
# PCL 1.12.0
# Eigen 3.4.0
# VTK 8.2.0
# C++ 14 （根据报错情况再修改你跑的项目的cmakelists.txt）
# jsoncpp，需要将jsoncpp里包含头文件的json文件夹拷贝到manual_calib的include文件夹中。（这个找到git的项目主页拉下来源码）

VTK的安装依赖比较麻烦，多看看教程

AndrewJSong, 2022-09-01:
> > 没错，使用docker尝试过在远程服务器/本机/虚拟机/有无显卡的机器上测试，均无法实现图形界面显示。最后还是自己安装PCL VTK OPENCV实现可视化界面
> 
> 你好，我在本地装了opencv2.4.13，pcl1.9.1但是lidar2camera和radar2camera的手动标定程序总是出错。你本地是怎么安装的啊

> > 没错，使用docker尝试过在远程服务器/本机/虚拟机/有无显卡的机器上测试，均无法实现图形界面显示。最后还是自己安装PCL VTK OPENCV实现可视化界面
> 
> 你好，我在本地装了opencv2.4.13，pcl1.9.1但是lidar2camera和radar2camera的手动标定程序总是出错。你本地是怎么安装的啊

opencv版本没有问题，问题出在其他的依赖上。本地安装依赖参考我上一个回复。lidar2camera的joint和Manuel都没有问题，radar我没试过，你可以试试看。古德拉克

AndrewJSong, 2022-09-01:
> > 没错，使用docker尝试过在远程服务器/本机/虚拟机/有无显卡的机器上测试，均无法实现图形界面显示。最后还是自己安装PCL VTK OPENCV实现可视化界面
> 
> 你好，我在本地装了opencv2.4.13，pcl1.9.1但是lidar2camera和radar2camera的手动标定程序总是出错。你本地是怎么安装的啊

opencv版本没有问题，问题出在其他的依赖上。本地安装依赖参考我上一个回复。lidar2camera的joint和Manuel都没有问题，radar我没试过，你可以试试看。古德拉克

NovemberChopin, 2022-09-03:
你好，关于运行手动标注segmentation fault报错已经解决了，在CalibrationInit和CalibrationScaleChange函数里面有一行代码：
modification_list_.[i]=tmp; ，改成 modification_list_.push_back(tmp);就可以了。

我觉得这行代码直接用 push_back会更好，用下标方式有时候就可以，有时候就不可以，环境来来回回装了好多次，太折磨人了

我的环境 ubuntu 18.04  opencv 2.4.13  pcl 1.9 eigen 3.3.8 pangolin 0.5

盐的甜
***@***.***

kuanghuijun, 2022-10-21:
> > Hi, To address this issue, I forked this repo and add a script for docker visualization. You could check: [https://github.com/BayRanger/SensorsCalibration](url) And the usage has been added to README
> > Bests, BR
> 
> Thanks! It's ok!

YhQIAO, 2023-03-19:
recommend that you should install ubuntu on a real machine , then run docker environment with nvidia docker container tools installed to shou gui

FlyingAnt2018, 2023-05-25:
> 

i have tried to show gui result on ubuntu desktop without nvidia GPU, failed.

laomeng0703, 2023-06-28:
> Hi, To address this issue, I forked this repo and add a script for docker visualization. You could check: [https://github.com/BayRanger/SensorsCalibration](url) And the usage has been added to README
> 
> Bests, BR

这个fork有用，我从这里拉取的仓库然后在root用户下编译文件和运行都没有问题。GUI也可以正常显示。

zouzhiwei217, 2023-07-10:
> 

请问一定需要GPU吗

AbdullahJirjees, 2024-06-13:
running the demo calibration tool from the local machine it worked successfully but from the docker container it is not working and showing the error 
Pangolin X11: Failed to open X display
I tried all the possible solutions but still not working any suggestion?

Kuntal491, 2024-12-27:
i have tried to show gui result on ubuntu desktop without nvidia GPU, failed. Is it mandatory have gpu to run this. I am trying to run the radar2camera and facing this issue. the display loads up but as soon as i press enter . it shows this error of Pangolin and windows closes

Kuntal491, 2025-01-02:
@BayRanger I tried by cloning this repository still the display issue with pangolin X11 exists.

EdwardBao1006, 2025-06-25:
请问我只需要标定毫米波雷达与相机，编译radar2camera成功了，但是选点以后按Enter就没反应了，也不进行手动标定界面了，这是怎么回事呢？