PJLab-ADG/SensorsCalibration #9: lidar2imu automatic Segmentation fault(core dumped)

chengih, 2022-06-13:
Run the test sample
[image]

Spring-lovely, 2022-06-16:
Hi,
I have updated the docker image v1 version, please use the command docker pull scllovewkf/ opencalib:v1 to pull .
Thank you!
[image]

Spring-lovely, 2022-06-16:
Hi,
I have updated the docker image v1 version, please use the command [docker pull scllovewkf/opencalib:v1](url) to pull .
Thank you!
[image]

chengih, 2022-06-16:
Thanks for the reply, it worked for me