PJLab-ADG/SensorsCalibration #11: lidar2camera/manual_calib 【Segmentation fault (core dumped)】

AndrewJSong, 2022-06-16:
Without docker, the environment is self configured, 
ubuntu18.04.6 pcl1.9.1 opencv 2.4.13.6 VTK 8.2.0 eigen3.3.9 pangolin 0.6
The compilation passes, but an error is reported when running:“**Segmentation fault (core dumped)**”
Is the display part required by additional configuration?
---
./bin/run_lidar2camera data/0.png data/0.pcd data/center_camera-intrinsic.json data/top_center_lidar-to-center_camera-extrinsic.json
data/center_camera-intrinsic.json
center_camera-intrinsic
intrinsic:
2109.75 0 949.828
0 2071.72 576.237
0 0 1
dist:
-0.108145 0.138668
top_center_lidar-to-center_camera-extrinsic
Loading data completed!
Segmentation fault (core dumped)

Spring-lovely, 2022-06-20:
Hi,
Please use docker image first，thanks!

daikunpeng, 2022-06-20:
> ## Without docker, the environment is self configured,
> ubuntu18.04.6 pcl1.9.1 opencv 2.4.13.6 VTK 8.2.0 eigen3.3.9 pangolin 0.6
> The compilation passes, but an error is reported when [...]

Hi,
Can you tell me how to obtain the center_camera-intrinsic.json and top_center_lidar-to-center_camera-extrinsic.json (in your case) files? The lidar to camera calibration must be carried out first place? What tools do you use?
Thank you.

AndrewJSong, 2022-06-20:
> SensorsCalibrationlidar2camera/manual_calib)/data 路径下有测试数据。