PJLab-ADG/SensorsCalibration #12: lidar2imu/manual_calib core dumped

smallzhao, 2022-06-18:
`gnss-to-top_center_lidar-extrinsic
1 0 0 0
0 1 0 0
0 0 1 0
0 0 0 1
load: 0/1080, data/top_center_lidar/2021-10-26-16-21-29-468.pcd1
1
Loading data completed!
=>Calibration scale Init!
run_lidar2imu: /build/pcl-OilVEB/pcl-1.8.1+dfsg1/octree/include/pcl/octree/impl/octree_pointcloud.hpp:688: void pcl::octree::OctreePointCloud<PointT, LeafContainerT, BranchContainerT, OctreeT>::genOctreeKeyforPoint(const PointT&, pcl::octree::OctreeKey&) const [with PointT = pcl::PointXYZI; LeafContainerT = pcl::octree::OctreeContainerPointIndices; BranchContainerT = pcl::octree::OctreeContainerEmpty; OctreeT = pcl::octree::OctreeBase<pcl::octree::OctreeContainerPointIndices, pcl::octree::OctreeContainerEmpty>]: Assertion `key_arg.x <= this->max_key_.x' failed.
已放弃 (核心已转储)`

Spring-lovely, 2022-06-20:
Hi,
I have updated the docker image v1 version, please use the command [docker pull scllovewkf/opencalib:v1](https://github.com/PJLab-ADG/SensorsCalibration/issues/url) to pull .
Thank you!

Spring-lovely, 2022-06-20:
For the appearance of the GUI graphics, please refer to this link[https://github.com/PJLab-ADG/SensorsCalibration/issues/7](url)

Super-Daryl, 2022-11-11:
I also encountered this problem .  how to solve it ?

Super-Daryl, 2022-11-11:
- **reason**:if octree empty ,use pcl isVoxelOccupiedAtPoint function will be core  dumped
- **solution**:  replace  run_lidar2imu.cpp file 375 line code with  below
`  double min_x, min_y, min_z, max_x, max_y, max_z;
      all_octree->getBoundingBox(min_x, min_y, min_z, max_x, max_y, max_z);
      bool isInBox = (dst_pt.x >= min_x && dst_pt.x <= max_x) && (dst_pt.y >= min_y && dst_pt.y <= max_y) && (dst_pt.z >= min_z && dst_pt.z <= max_z);
      if (!isInBox || !all_octree->isVoxelOccupiedAtPoint(dst_pt))
      {
        all_octree->addPointToCloud(dst_pt, cloudLidar);
      }`