PJLab-ADG/SensorsCalibration #16: Question about LiDAR-IMU calibration tool

Taeyoung96, 2022-06-27:
@Spring-lovely  @xiaokn   
Hi, Thanks for your great work!  

I have some questions about the `lidar2imu` tool.  

1. How could I get `lidar_pose_file` to use this tool?  
2. What tool did you use to get the `lidar_pose_file` in your experiment?  
3. Does initial values of extrinsic parameters have a lot of effect on the result?  

Thanks,

qyj1996, 2022-07-03:
> @Spring可爱的@xiaokn 嗨，谢谢你的出色工作！
> 
> 我对该工具有一些疑问。`lidar2imu`
> 
> 1. 我怎么能使用这个工具？`lidar_pose_file`
> 2. 您使用什么工具在实验中获取了 ？`lidar_pose_file`
> 3. 外在参数的初始值对结果有很大影响吗？
> 
> 谢谢

Hello, I have the same problem with you. The initial external parameters seem to be closely related to the calibration results. The initial value of external participation after calibration is not much different. I try to change the external parameters slightly, and the calibration will not be able to proceed. How do you solve this problem.

xiaokn, 2022-07-04:
> @Spring-lovely @xiaokn Hi, Thanks for your great work!
>
> I have some questions about the `lidar2imu` tool.
>
> 1. How could I get `lidar_pose_file` to use this tool?
> 2. What tool did you use to get the [...]

1. lidar_pose_file is the pose output by the inertial navigation device after transformation with lidar-imu extrinsic parameters.
2. The initial extrinsic parameters have little effect on it, and the calibration environment may impact the results.

Taeyoung96, 2022-07-04:
@xiaokn   
Thanks for your reply.  
Could you recommend some tools that make `lidar_pose_file`?  

Thanks,

closed 2024-05-24 by Taeyoung96