PJLab-ADG/SensorsCalibration #17: Lidar

xiahaa, 2022-06-29:


xiaokn, 2022-07-04:
> hi, thanks for sharing this repository. however, i do find the code of lidar2camera calibration does not match with what has been described in your paper.
> 
> 1. p3l is mentioned to be used for coarse calibration, while i did not find any code related to p3l;
> 2. idt is mentioned to be used for cost formulation, while i did not find any code related to idt;
> 
> i tried to perturb the extrinsics a little bit, say +/- 1 degree. the implementation cannot give me any improvements, actually, the extrinsics stays the same before and after calibration.
> 
> is it the real implementation mentioned in your paper?

Sorry, for some reason, I deleted this part of the code and will add it later.

HuangVictorAuto, 2022-08-04:
@xiaokn , auto lidar2camera calibration, we have to prepare to mask the images, but when i setup the conda env as required and run the merge_mask.py, the program will be stuck and no masked image will output. Is there any step, that i have missed?

russellyq, 2022-09-23:
Hi @xiaokn and @HuangVictorAuto 

I meet the same problem. Did you solve it ?

Muxindawang, 2023-05-30:
> Sorry, for some reason, I deleted this part of the code and will add it later.

> Sorry, for some reason, I deleted this part of the code and will add it later.

Therefore, when will it be added? I also encountered the same problem in use, which could not be corrected after making a deviation of 1 degree.

zouzhiwei217, 2023-07-27:
> @xiaokn , auto lidar2camera calibration, we have to prepare to mask the images, but when i setup the conda env as required and run the merge_mask.py, the program will be stuck and no masked image will output. Is there any step, that i have missed?，自动Lidar2Camera校准，我们必须准备屏蔽图像，但是当我根据需要设置Conda Env并运行merge_mask.py时，程序将卡住并且不会输出遮罩图像。有什么步骤，我错过了吗？

我也遇到了,您解决了吗.

iegorval, 2025-05-15:
@xiaokn I also find it very confusing that the paper stated that you compute the coarse calibration from the data itself, while the code in this repo expects some initial calibration guess. Is that correct? Is it really based on the article?