PJLab-ADG/SensorsCalibration #21: lidar2imu

qyj1996, 2022-07-03:
Hello, author, first of all, thank you for providing such a good toolbox, but I have some questions to ask you. In lidar2imu tool, the initial external parameters seem to be closely related to the calibration results, and the initial value of external participation after calibration is not much different. I try to change the external parameters slightly, and the calibration will not be able to proceed. Is this normal? But the actual situation is that we cannot obtain such an accurate initial external parameter through measurement.

xiaokn, 2022-07-04:
Thank you very much for your question. I found that there are indeed some problems here. We will fix it later, if you need, you can send me an email, and I can send you another version of our lidar- imu calibration tool.

qyj1996, 2022-07-04:
> 非常感谢您的提问。我发现这里确实存在一些问题。我们稍后会修复它，如果需要，您可以给我发送电子邮件，我可以向您发送另一个版本的激光雷达 - imu校准工具。

Thank you very much for your reply. My email is 453225147@qq.com , thank you very much for providing a new lidar2imu calibration tool, thank you.

qyj1996, 2022-07-04:
> Thank you very much for your question. I found that there are indeed some problems here. We will fix it later, if you need, you can send me an email, and I can send you another version of our lidar- imu calibration tool.

In addition, for the manual calibration tool in this project, if I use the initial external parameters in the data set provided by you, the initial point cloud calibration effect displayed on the interface is very poor, and after I modify the initial external parameters to the unit matrix, the initial point cloud calibration effect displayed on the interface is very good. If I change the translation and rotation slightly, the point cloud image will become worse. What is the reason? I also hope you can give me some guidance.

qyj1996, 2022-07-04:
Thank you very much for your reply. My email is 453225147@qq.com, thank you very much for providing a new lidar2imu calibration tool, thank you.

w-shuw, 2023-07-06:
我也遇到了将初始外参改为单位阵反而点云对齐效果比较好的问题，请问知道这是什么原因了吗