PJLab-ADG/SensorsCalibration #22: Lidar2imu custom data problem on auto_calib

CurssedCoffin, 2022-07-05:
I'm using my own pcd data and the head is like this
```
# .PCD v0.7 - Point Cloud Data file format
VERSION 0.7
FIELDS x y z intensity
SIZE 4 4 4 1
TYPE F F F U
COUNT 1 1 1 1
WIDTH 1
HEIGHT 66243
VIEWPOINT 0 0 0 1 0 0 0
POINTS 66243
DATA binary
```

The intensity is uint8_t and I changed the custom pointcloud structure in ```auto_calib/include/common/Lidar_parser_base.h```
```
struct LidarPointXYZIRT {
  PCL_ADD_POINT4D;
  uint8_t intensity;
  EIGEN_MAKE_ALIGNED_OPERATOR_NEW
} EIGEN_ALIGN16;

POINT_CLOUD_REGISTER_POINT_STRUCT(
    LidarPointXYZIRT,
    (float, x, x)(float, y, y)(float, z, z)(uint8_t, intensity, intensity))
```

But when i'm running ```run_lidar2imu``` I got warnings showing ```failed to find match for field 'intensity'```

And the ceres iteration also shows like
```
==>ROUND 22
deltaQ: 1 0 0 0
solver.cc:415 Function tolerance reached. No non-constant parameter blocks found.
Ceres Solver Report: Iterations: -2, Initial cost: 0.000000e+00, Final cost: 0.000000e+00, Termination: CONVERGENCE
delta rpy: 0 0 0
delta T: 0 0 0
Round Finish!
```

How to solve this problem?

Thanks!

CurssedCoffin, 2022-07-05:
The pcd data is extracted from rosbag by the way

CurssedCoffin, 2022-07-06:
I have solved this problem by debugging.

closed 2022-07-06 by CurssedCoffin

luameows, 2023-03-16:
@CurssedCoffin did u try their auto_calib? What about the result?

CurssedCoffin, 2023-03-17:
> @CurssedCoffin did u try their auto_calib? What about the result?

The Lidar-IMU auto calib result is not good on my dataset of Robosense RSBP lidar and a GPS+IMU device. The extrinsic matrix is not so accurate. When I use a 1 min data of rosbag, and project lidar frames to one single frame, it's not closed and have some edges on it.

pedestrain123, 2024-10-14:
@CurssedCoffin Hello, I used my own dataset to test the calibration code of Auto Calib, and the calibration results were not satisfactory. However, the author's open-source data showed good results. Could you please open up your data package for me to test？

Milankaka2222, 2025-03-18:
> I'm using my own pcd data and the head is like this
>
> ```
> # .PCD v0.7 - Point Cloud Data file format
> VERSION 0.7
> FIELDS x y z intensity
> SIZE 4 4 4 1
> TYPE F F F U
> COUNT 1 1 1 1
> WIDTH 1
> HEIGHT 66243 [...]

Can you please advise the blogger on how to solve this issue？？？thanks

CurssedCoffin, 2025-03-18:
> Can you please advise the blogger on how to solve this issue？？？thanks

change ```lidar2imu/auto_calib/include/common/Lidar_parser_base.h```

FROM
```cpp
struct LidarPointXYZIRT {
  PCL_ADD_POINT4D;
  float intensity;
  uint16_t ring;
  double timestamp;
  EIGEN_MAKE_ALIGNED_OPERATOR_NEW
} EIGEN_ALIGN16;

POINT_CLOUD_REGISTER_POINT_STRUCT(
    LidarPointXYZIRT,
    (float, x, x)(float, y, y)(float, z, z)(float, intensity, intensity)(
        uint16_t, ring, ring)(double, timestamp, timestamp))
```

TO
```cpp
struct LidarPointXYZIRT {
  PCL_ADD_POINT4D;
  uint8_t intensity;
  EIGEN_MAKE_ALIGNED_OPERATOR_NEW
} EIGEN_ALIGN16;

POINT_CLOUD_REGISTER_POINT_STRUCT(
    LidarPointXYZIRT,
    (float, x, x)(float, y, y)(float, z, z)(uint8_t, intensity, intensity))
```

**OR** do global search and replace:

FROM
```cpp
pcl::PointCloud<LidarPointXYZIRT>::Ptr
```

TO
```cpp
pcl::PointCloud<pcl::PointXYZI>::Ptr
```

may help fix this warning, it's so long ago since I'm testing this repo, and this can also be not correct, sorry.