PJLab-ADG/SensorsCalibration #27: lidar2imu

qyj1996, 2022-07-28:
Hello, author, first of all, thank you for providing such a good toolbox, but I have some questions to ask you. In lidar2imu tool,is the setting of the initial external parameter the external parameter of the conversion from lidar system to IMU system, or the external parameter of the conversion from IMU system to lidar system? Take the translation as an example. Please tell me in detail whether it is the translation amount set by lidar plus IMU to convert to the IMU coordinate system, or the translation amount set by IMU plus lidar to convert to the lidar coordinate system?

jiahao68, 2022-08-01:
The setting of the initial external parametethe is the conversion from IMU system to lidar system.

jiahao68, 2022-08-01:
> Hello, author, first of all, thank you for providing such a good toolbox, but I have some questions to ask you. In lidar2imu tool,is the setting of the initial external parameter the external [...]

The setting of the initial external parametethe is the conversion from IMU system to lidar system.

qyj1996, 2022-08-01:
> The setting of the initial external parametethe is the conversion from IMU system to lidar system.

Is IMU the parent coordinate system and lidar the child coordinate system?

jiahao68, 2022-08-01:
> Is IMU the parent coordinate system and lidar the child coordinate system?

Yes

qyj1996, 2022-08-01:
> Yes

But in the actual setting, I set IMU as the child coordinate system and lidar as the parent coordinate system to calibrate. What is the reason?

xiaokn, 2022-08-01:
> But in the actual setting, I set IMU as the child coordinate system and lidar as the parent coordinate system to calibrate. What is the reason?

You may need to use the inverse matrix of the extrinsic parameter when stitching LiDAR point clouds.

xiaokn, 2022-08-01:
> But in the actual setting, I set IMU as the child coordinate system and lidar as the parent coordinate system to calibrate. What is the reason?

No matter which setting, as long as the LiDAR point cloud can be stitched into a local map

qyj1996, 2022-08-01:
> Referenc

OK, thank you very much for your reply.