PJLab-ADG/SensorsCalibration #28: 激光雷达和IMU外参标定

2017DEVIL, 2022-08-14:
请问在标定激光雷达和IMU的外参时，要求确保地面足够平坦，这种情况下IMU在Z轴方向的激励不够充分，那么雷达和IMU在z轴方向的外参标定效果会不会不好呢？

xiaokn, 2022-08-15:
> 请问在标定激光雷达和IMU的外参时，要求确保地面足够平坦，这种情况下IMU在Z轴方向的激励不够充分，那么雷达和IMU在z轴方向的外参标定效果会不会不好呢？

你好，感谢您提出的问题。(我们上传的这一个版本lidar2imu有点bug，后面会再更新一版，另外这里起名imu也有点不妥，应该是lidar到pose-sensor的外参标定，后续也会修改一下哈)。您提到的平面运动确实会影响z轴的标定，所以这里我们直接用测量值作为z轴标定的结果。不过，我们后面更新的版本也会解决因为平面运动z轴无法标定的问题。

liang0724s, 2022-08-29:
> > 请问在标定激光雷达和IMU的外参时，要求确保地面足够平坦，这种情况下IMU在Z轴方向的激励不够充分，那么雷达和IMU在z轴方向的外参标定效果会不会不好呢？
> 
> 你好，感谢您提出的问题。(我们上传的这一个版本lidar2imu有点bug，后面会再更新一版，另外这里起名imu也有点不妥，应该是lidar到pose-sensor的外参标定，后续也会修改一下哈)。您提到的平面运动确实会影响z轴的标定，所以这里我们直接用测量值作为z轴标定的结果。不过，我们后面更新的版本也会解决因为平面运动z轴无法标定的问题。

您这里提到是lidar到pose-sensor的外参标定，那标定工具的输入项“lidar_pose_file: Poses file from the IMU sensor”，可以把它理解为imu的pose经过利用lidar-imu初始外参转换后的pose值吗？

lingbo-yu, 2022-12-14:
@liang0724s 你好，麻烦请问一下这个lidar pose file确实是rtk的pose经过初始外参转换过后的吗？

liang0724s, 2022-12-21:
> @liang0724s 你好，麻烦请问一下这个lidar pose file确实是rtk的pose经过初始外参转换过后的吗？
你好，你可以从函数ProcessLidarFrame的点云坐标系转换过程中理解这个lidar pose file.