PJLab-ADG/SensorsCalibration #30: radar2cmara calibration

wang6342, 2022-08-23:
您好
    非常感谢你们对多传感器标定做出的相关工作，关于毫米波雷达和相机标定，有几个问题想请教一下。
    1.我研读了你们论文毫米波和相机标定的相关部分，也配置好了相关算法，好像没有提到毫米波和相机标定结果怎么评价，是首先毫米波数据和相机数据在鸟瞰图上显示，再就是相关数据点和参考的车道线平行就认为标定是准确的吗
    2.关于标定数据采集。radar CSV格式的数据可以用ROS录得bag直接转换过来吗，标定对于毫米波雷达来说，主要是关注csv数据里面velocity_x, velocity_y, position_x，position_y值吗，dynprop，rcs(反射截面积)分别指的是什么？后面其他的值怎么给出，能方便给个制作这种radar数据的教程吗。
   3.毫米波雷达和相机初始外参是怎么得到的，是我大概给一个，再在界面上手动调整就可以吗

xiaokn, 2022-08-24:
> 您好 非常感谢你们对多传感器标定做出的相关工作，关于毫米波雷达和相机标定，有几个问题想请教一下。 1.我研读了你们论文毫米波和相机标定的相关部分，也配置好了相关算法，好像没有提到毫米波和相机标定结果怎么评价，是首先毫米波数据和相机数据在鸟瞰图上显示，再就是相关数据点和参考的车道线平行就认为标定是准确的吗 2.关于标定数据采集。radar [...]

手动标定工具，主要有两个参考：1. radar本身的点云投影到图像上，但是因为Radar的稀疏性和二维点不可能像雷达点云那样准确判断 2. 为了评估yaw角需要把图像和radar都转到鸟瞰图上判断，这里需要标定一下图像到地面的H矩阵

xiaokn, 2022-08-24:
> 您好 非常感谢你们对多传感器标定做出的相关工作，关于毫米波雷达和相机标定，有几个问题想请教一下。 1.我研读了你们论文毫米波和相机标定的相关部分，也配置好了相关算法，好像没有提到毫米波和相机标定结果怎么评价，是首先毫米波数据和相机数据在鸟瞰图上显示，再就是相关数据点和参考的车道线平行就认为标定是准确的吗 2.关于标定数据采集。radar [...]

2. radar数据的处理可能要根据具体的radar来看，只要能得到radar点的坐标就可以了 3. 毫米波雷达和相机的初始外参可以根据传感器的坐标轴确定，一般是单位矩阵进行（或者转90°这样子）4. 这里有个比较重要的需要标定一下相机的H矩阵才能把相机转到鸟瞰图

wang6342, 2022-08-24:
感谢您的回复！
1.评估yaw角意思就是把camera和radar数据都投影到了X0Z平面吗，那我在上位机鸟瞰图上看到的是radar的点和车道线平行就认为标定准确了吗还是radar点和两边草丛重合对齐就标定准确啦。
2.我看你们的程序第一步已经在用四点法在求H矩阵了，这个需要我再单独先求出来个结果吗？第一步求出H是直接写到jason文件吗，我怎么看好像没有写进去。

我再研究研究，谢谢您。

alfanrizaldy, 2023-02-22:
Hello @wang6342 
I'm facing difficulties about your question number 3, how you decide the external parameter between camera and radar?
Thank you

idiotcv, 2024-08-31:
您好，计算相机到地面的单应矩阵中用到的center_camera-homography.json，数据是怎么来的呢？

pedestrain123, 2024-10-28:
> 您好 非常感谢你们对多传感器标定做出的相关工作，关于毫米波雷达和相机标定，有几个问题想请教一下。 1.我研读了你们论文毫米波和相机标定的相关部分，也配置好了相关算法，好像没有提到毫米波和相机标定结果怎么评价，是首先毫米波数据和相机数据在鸟瞰图上显示，再就是相关数据点和参考的车道线平行就认为标定是准确的吗 2.关于标定数据采集。radar [...]

@wang6342 @xiaokn 请问velocity_x, velocity_y, position_x，position_y分别指的是毫米波雷达坐标系的xy方向的速度和位置坐标吗？