PJLab-ADG/SensorsCalibration #31: lidar2imu auto_calib (segmentation fault)

zengwen2017, 2022-08-23:
作者你好,感谢你提供的标定工具，在进行lidar和imu的自动标定时，出现了segmentation fault报错。
看到您在其它问题下面的回答，尝试了docker pull images 运行代码还是出现了问题，请问还有其他的解决方法吗？如果可以的话，请发送到我的邮箱：zengwen2017@qq.com

xiaokn, 2022-08-24:
> 作者你好,感谢你提供的标定工具，在进行lidar和imu的自动标定时，出现了segmentation fault报错。 看到您在其它问题下面的回答，尝试了docker pull images 运行代码还是出现了问题，请问还有其他的解决方法吗？如果可以的话，请发送到我的邮箱：[zengwen2017@qq.com](mailto:zengwen2017@qq.com)

我更新了一下lidar2imu的auto_calib的代码，你重新拉一下代码试试哈

zengwen2017, 2022-08-25:
作者你好，我尝试了您最新版本的代码，仍然出现了segmentation fault的报错，不知道该怎么解决，希望能获得帮助，感谢。

Lin239522, 2022-08-27:
你好，请问你其他代码运行时出现了类似问题吗？我在/lidar2lidar/manual_calib的代码运行中也遇到“段错误 (核心已转储)”的问题
[image]

zengwen2017, 2022-08-29:
作者你好，在运行autocalib部分代码时，检测到34轮时，出现了octree方面的报错：Assertion ‘cloud_arg==input_’ failed错误请问应该怎么解决

zengwen2017, 2022-09-02:
[image]
作者你好，我在使用auto_calibration是出现这种情况，请问是什么愿原因呢？

NovemberChopin, 2022-09-03:
> 作者你好，我尝试了您最新版本的代码，仍然出现了segmentation fault的报错，不知道该怎么解决，希望能获得帮助，感谢。
> […](#)

关于手动标注segmentation fault报错，我是这么解决的：在CalibrationInit和CalibrationScaleChange函数里面有一行代码：
`modification_list_.[i]=tmp;` ，改成 `modification_list_.push_back(tmp);`就可以运行了。
我觉得既然都用vector了，还是直接push_back操作安全一些，用下标赋值有时候会有问题的

zengwen2017, 2022-09-07:
作者你好，请问你在编写自动标定的代码时，用于测试的激光雷达类型是固态的还是机械的？有考虑过两者之间的差别吗？

zengwen2017, 2022-10-15:
作者你好，有一些问题请教。
radar2lidar手动标定代码中radar需要的csv文件中各数据应该怎么获得呢？有什么推荐的工具吗？第一列time_ns是什么含义，为什么值都一样？

huangwefeng121, 2023-06-16:
> 作者你好，请问你在编写自动标定的代码时，用于测试的激光雷达类型是固态的还是机械的？有考虑过两者之间的差别吗？

我也想知道lidar2imu对360度机械雷达和视野范围有限的固态雷达都适用吗？博主或者 @zengwen2017 你们对这个有什么建议吗？

aifeixingdelv, 2023-11-15:
> > 作者你好，我尝试了您最新版本的代码，仍然出现了segmentation fault的报错，不知道该怎么解决，希望能获得帮助，感谢。
> > […](#)
> 
> 关于手动标注segmentation fault报错，我是这么解决的：在CalibrationInit和CalibrationScaleChange函数里面有一行代码： `modification_list_.[i]=tmp;` ，改成 `modification_list_.push_back(tmp);`就可以运行了。 我觉得既然都用vector了，还是直接push_back操作安全一些，用下标赋值有时候会有问题的

Thank you !!! The method is very useful!!!

pedestrain123, 2024-10-15:
@huangwefeng121 你好请问你有用过opencalib里面的lidar2imu的auto calib方法吗？