PJLab-ADG/SensorsCalibration #35: lidar2imu diverge

linwk20, 2022-09-12:
Hi, Thanks for your sharing!
I have used the lidar2imu tool, however, the result seems to diverge.

`Ceres Solver Report: Iterations: 15, Initial cost: 2.748471e+03, Final cost: 2.717503e+03, Termination: CONVERGENCE`
 
I have given it a proper initialization. What could be the possible other reason?

Thanks!

xiaokn, 2022-09-13:
> Hi, Thanks for your sharing! I have used the lidar2imu tool, however, the result seems to diverge.
>
> `Ceres Solver Report: Iterations: 15, Initial cost: 2.748471e+03, Final cost: 2.717503e+03, [...]

You can pull the latest code and try again. It is better to have more surface features in the calibration environment.

linwk20, 2022-09-13:
Hi, Thanks for your quick reply, I will try running the latest code.
Besides, I wonder is there other built-in debug way? For example, can I visualize the diverging process or the surface feature's distribution? now I can only received feedback from terminal.

pedestrain123, 2024-10-14:
@linwk20 您好，请问你用自己的数据auto calib标定lidar2imu标定成功了吗？我用自己的数据集标定效果不理想，可以交流一下吗？期待您的回复

linwk20, 2024-10-14:
no, i use other tools

pedestrain123, 2024-10-14:
@linwk20 请问您用的是什么工具，lidar align吗？还是什么其他工具？可以推荐一下吗？能否加你微信或者QQ交流一下，QQ2985782675,v:19516191665

linwk20, 2024-10-14:
Actually, other people take care of the calibration for me, I am not sure what they use too. The result is not good as well.

closed 2024-10-14 by linwk20

pedestrain123, 2024-10-14:
> 实际上，其他人帮我做了校准，我不确定他们用什么。结果也不好。我建议你使用其他开源校准工具。或者在更多数据集上尝试这个 repo，也许有些会起作用。

谢谢