PJLab-ADG/SensorsCalibration #37: Pangolin X11: Unable to retrieve framebuffer options

leejoonho95, 2022-09-13:
terminate called after throwing an instance of 'std::runtime_error'
  what():  Pangolin X11: Unable to retrieve framebuffer options
Aborted (core dumped)

xiaokn, 2022-09-19:
> terminate called after throwing an instance of 'std::runtime_error' what(): Pangolin X11: Unable to retrieve framebuffer options Aborted (core dumped)

You can refer to this issue: https://github.com/PJLab-ADG/SensorsCalibration/issues/7

Aiuan, 2022-11-23:
看起来这个问题并没有得到解决，使用docker，在利用lidar2imu包中的manua_calib包，运行./bin/run_lidar2imu ./data/top_center_lidar/ ./data/top_center_lidar-pose.txt ./data/gnss-to-top_center_lidar-extrinsic.json
还是会报出以下错误：
terminate called after throwing an instance of 'std::runtime_error'
  what():  Pangolin X11: Unable to retrieve framebuffer options
Aborted (core dumped)

zhuoyuan00, 2022-12-12:
yes this problem occurs when using offical docker

EmilioOlivastri, 2023-01-16:
Has anyone solved this error? I tried #7 but it didn't fix the problem.

Kuan-Wei-Tseng, 2023-05-01:
This problem still occurs when I tried to launch the code inside official docker on my Ubuntu 20.04 virtual machine. I built the code directly without using official docker to circumvent this problem.

HakanTuncel42, 2023-06-28:
I use Ubuntu 20.04 Pangolin v0.6 and when i try lidar2imu manual calib it only loads one of the PCD files ... do you know how i can fix this ?