PJLab-ADG/SensorsCalibration #38: 关于lider2camera标定的auto模式

wo712268, 2022-09-15:
你好！感谢分享代码！我在自己电脑上进行了尝试，尝试之前我将外参数据改为了距离正确值比较近的值。
但是用example跑完以后，发现误差没有减少，反而增多了，新计算出的标定数据距离真值远了些，我注意到论文里提到的，auto标定这里计算了一系列的道路数据，所以这个标定工具应该是不止一张图才能使用的吗？

xiaokn, 2022-09-19:
> 你好！感谢分享代码！我在自己电脑上进行了尝试，尝试之前我将外参数据改为了距离正确值比较近的值。 但是用example跑完以后，发现误差没有减少，反而增多了，新计算出的标定数据距离真值远了些，我注意到论文里提到的，auto标定这里计算了一系列的道路数据，所以这个标定工具应该是不止一张图才能使用的吗？

因为特征分割可能会存在一些误差，所以精度会受到影响，多帧经过后处理会有一些提升。后续我们会继续优化lidar2camera auto的标定工具!

callzhang, 2022-11-23:
请问多帧处理如何做？

zhangyuran-gg, 2024-11-25:
> > 你好！感谢分享代码！我在自己电脑上进行了尝试，尝试之前我将外参数据改为了距离正确值比较近的值。但是用example跑完以后，发现误差没有减少，反而增多了，新计算出的标定数据距离真值远了些，我注意到论文里提到的，auto标定这里计算了一系列的道路数据，所以这个标定工具应该是不止一张图才能使用的吗？
> 
> 因为特征分割可能会存在一些误差，所以精度会受到影响，多帧经过后处理会有一些提升。后续我们会继续优化lidar2camera auto的标定工具！

你好，我使用自己的数据来测试auto_calib_v2，但是总是无法计算出结果，这是为什么？输出如下图
[image]

zhangyuran-gg, 2024-11-25:
> 你好，我使用自己的数据来测试auto_calib_v2，但是总是无法计算出结果，这是为什么？输出如下图 [...]

我调整了初始外参并重新运行程序，它成功运行并给出了标定结果，但是.png看着误差还是比较大，看输出只检查到一个plane，对此有什么建议吗？
[image]