PJLab-ADG/SensorsCalibration #40: Is there any difference between auto calibration and online calibration w.r.t lidar2imu?

zhh2005757, 2022-10-15:
It seems to me that both auto calibration methods and online methods of lidar2imu need lidar pcds and imu poses input. What the difference between them? I think online calibration can run without any input, right?

xiaokn, 2022-10-17:
> It seems to me that both auto calibration methods and online methods of lidar2imu need lidar pcds and imu poses input. What the difference between them? I think online calibration can run without any input, right?

Automatic calibration requires users to collect calibration data as required, while online calibration does not. Both of these require sensor data input.

zhh2005757, 2022-10-17:
> 

But I see that the online calibration tool also need some sensor data input files as the starting parameters. Are theses input files the same as ones of auto calibration?