PJLab-ADG/SensorsCalibration #45: Definition of parameter json?

callzhang, 2022-11-22:
```json
{
    "center_camera-intrinsic": {
        "sensor_name": "center_camera",
        "target_sensor_name": "center_camera",
        "device_type": "camera",
        "param_type": "intrinsic",
        "param": {
            "img_dist_w": 1920,
            "img_dist_h": 1200,
            "cam_K": {
                "rows": 3, # what is this?
                "cols": 3, # what is this?
                "type": 6, # what is this?
                "continuous": true, # what is this?
                "data": [
                    [
                        2117.31,
                        0,
                        924.681
                    ],
                    [
                        0,
                        2113.29,
                        656.457
                    ],
                    [
                        0,
                        0,
                        1.0
                    ]
                ]
            },
            "cam_dist": { # how to define camera model?
                "rows": 1,  # what is this?
                "cols": 5, # what is this?
                "type": 6, # what is this?
                "continuous": true, # what is this?
                "data": [ # (k1,k2,p1,p2[,k3[,k4,k5,k6[,s1,s2,s3,s4[,τx,τy]]]])?
                    [
                        -0.102933,
                        -0.040925,
                        0.00057951,
                        -0.00419933,
                        0.429959
                    ]
                ]
            }
        }
    }
}
```

```
{
    "top_center_lidar-to-center_camera-extrinsic": {
        "sensor_name": "top_center_lidar",
        "target_sensor_name": "center_camera", 
        "device_type": "relational",  # what is this?
        "param_type": "extrinsic",
        "param": {
            "time_lag": 0,
            "sensor_calib": {
                "rows": 4,
                "cols": 4,
                "type": 6,
                "continuous": true,
                "data": [
                    [
                        0.00382471,
                        -0.999992,
                        -0.00070554,
                        -0.0125114
                    ],
                    [
                        -0.0132276,
                        0.000654817,
                        -0.999912,
                        -0.379526
                    ],
                    [
                        0.999905,
                        0.00383377,
                        -0.0132251,
                        -0.551037
                    ],
                    [
                        0,
                        0,
                        0,
                        1
                    ]
                ]
            }
        }
    }
}
```

xiaokn, 2022-12-06:
> ```json
> {
>     "center_camera-intrinsic": { # what's the naming schema?
>         "sensor_name": "center_camera",
>         "target_sensor_name": "center_camera",
>         "device_type": "camera",
>         "param_type": "intrinsic",
>         "param": {
>             "img_dist_w": 1920,
>             "img_dist_h": 1200,
>             "cam_K": {
>                 "rows": 3, # what is this?
>                 "cols": 3, # what is this?
>                 "type": 6, # what is this?
>                 "continuous": true, # what is this?
>                 "data": [
>                     [
>                         2117.31,
>                         0,
>                         924.681
>                     ],
>                     [
>                         0,
>                         2113.29,
>                         656.457
>                     ],
>                     [
>                         0,
>                         0,
>                         1.0
>                     ]
>                 ]
>             },
>             "cam_dist": { # how to define camera model? i.e. how to define camera model other than pin hole?
>                 "rows": 1,  # what is this?
>                 "cols": 5, # what is this?
>                 "type": 6, # what is this?
>                 "continuous": true, # what is this?
>                 "data": [ # (k1,k2,p1,p2[,k3[,k4,k5,k6[,s1,s2,s3,s4[,τx,τy]]]])?
>                     [
>                         -0.102933,
>                         -0.040925,
>                         0.00057951,
>                         -0.00419933,
>                         0.429959
>                     ]
>                 ]
>             }
>         }
>     }
> }
> ```
> 
> ```json
> {
>     "top_center_lidar-to-center_camera-extrinsic": { # what's the naming schema?
>         "sensor_name": "top_center_lidar",
>         "target_sensor_name": "center_camera", 
>         "device_type": "relational",  # what is this?
>         "param_type": "extrinsic",
>         "param": {
>             "time_lag": 0,
>             "sensor_calib": {
>                 "rows": 4,
>                 "cols": 4,
>                 "type": 6,
>                 "continuous": true,
>                 "data": [
>                     [
>                         0.00382471,
>                         -0.999992,
>                         -0.00070554,
>                         -0.0125114
>                     ],
>                     [
>                         -0.0132276,
>                         0.000654817,
>                         -0.999912,
>                         -0.379526
>                     ],
>                     [
>                         0.999905,
>                         0.00383377,
>                         -0.0132251,
>                         -0.551037
>                     ],
>                     [
>                         0,
>                         0,
>                         0,
>                         1
>                     ]
>                 ]
>             }
>         }
>     }
> }
> ```
This template can be modified according to your needs, many parameters are not used here

callzhang, 2022-12-06:
My question was the definition of the param. Please see questions marked inside. Thanks!