PJLab-ADG/SensorsCalibration #52: cann't open SensorX2car

Tausc, 2023-01-13:
Hello, thank you for sharing this repository!
I'm interested in the online calibration of sensor-to-car coordinate systems, but when I try to open the link given in SensorX2car, it shows that page not found. 
Can you provide another way to get the related code? Thank you very much!

fabioreway, 2023-01-16:
I'm also interested. Thank you.

xiaokn, 2023-01-18:
> Hello, thank you for sharing this repository! I'm interested in the online calibration of sensor-to-car coordinate systems, but when I try to open the link given in SensorX2car, it shows that page not found. Can you provide another way to get the related code? Thank you very much!

Thanks for your question, the link is now open.

xiaokn, 2023-01-18:
> I'm also interested. Thank you.

Thanks for your question, the link is now open

closed 2023-04-30 by Tausc