PJLab-ADG/SensorsCalibration #54: How to get the initial value of external parameters between radar and camera?

alfanrizaldy, 2023-02-15:
https://github.com/PJLab-ADG/SensorsCalibration/issues/3
I've see the answer in above link but still no idea how to find that
Can you slightly detail answering this question?
Thanks

xiaokn, 2023-02-23:
> #3 I've see the answer in above link but still no idea how to obtain that Can you slightly detail answering this question? Thanks

For example, the translation can be measured, and the rotation is roughly corresponding to the coordinate axis of the installed sensor.

xiaokn, 2023-02-23:
> So, it is manually metered? I got the idea for the rotation, so i measure how the facing angle of the radar Then, how to measure the translation? I mean, with what it compared or referred?

For example, use a ruler to measure manually or give the design value according to the CAD drawing.

alfanrizaldy, 2023-02-23:
[image]
I've made simple illustration above, then i measure the between the radar and what?

closed 2023-05-12 by alfanrizaldy