PJLab-ADG/SensorsCalibration #55: joint_calib  csv文件

Muxindawang, 2023-02-15:
[image]
请问大佬 csv文件中 点的顺序是左上  右上， 请问为什么左上的点的y值比右上的点的y值小呢？lidar不是右手坐标系吗？正常右手坐标系，y轴指向左侧，那么左上点的y值应该比右上角的y值大才对呀，求大佬解惑，非常感谢。

xiaokn, 2023-02-23:
> lidar不是右手坐标系吗？正常右手坐标系，y轴指向左侧，那么左上点的y值应该比右上角的y值大才对呀，

carla是左手坐标系，里面Y轴是朝右边的

Muxindawang, 2023-02-23:
> carla

好的 多谢 请问您有用这个方法标定过实际的场景吗 我最近标定实际场景 发现效果比较一般 当然这和我采集的数据应该有一定的关系

Muxindawang, 2023-02-23:
> > carla
> 
> 好的 多谢 请问您有用这个方法标定过实际的场景吗 我最近标定实际场景 发现效果比较一般 当然这和我采集的数据应该有一定的关系

我采集的数据 大概距离车3m到4m左右，  请问用这样的数据标定之后 远距离的精度是不是会降低，误差随着距离的增加进一步被放大？

xiaokn, 2023-02-23:
> > > carla
> > 
> > 
> > 好的 多谢 请问您有用这个方法标定过实际的场景吗 我最近标定实际场景 发现效果比较一般 当然这和我采集的数据应该有一定的关系
> 
> 我采集的数据 大概距离车3m到4m左右， 请问用这样的数据标定之后 远距离的精度是不是会降低，误差随着距离的增加进一步被放大？

有些公司用的实际数据上标定精度还挺高的，具体可能也要看情况，比如传感器，标定板，数据质量等。一般来说如果有远处的标志物可能更好一点，但是标定间一般空间有限，如果能保证近处足够准，远处一般也相对来说是准的。

Muxindawang, 2023-02-24:
> > > > carla
> > > 
> > > 
> > > 好的 多谢 请问您有用这个方法标定过实际的场景吗 我最近标定实际场景 发现效果比较一般 当然这和我采集的数据应该有一定的关系
> > 
> > 
> > 我采集的数据 大概距离车3m到4m左右， 请问用这样的数据标定之后 远距离的精度是不是会降低，误差随着距离的增加进一步被放大？
> 
> 有些公司用的实际数据上标定精度还挺高的，具体可能也要看情况，比如传感器，标定板，数据质量等。一般来说如果有远处的标志物可能更好一点，但是标定间一般空间有限，如果能保证近处足够准，远处一般也相对来说是准的。

远处的话 对于标定板尺寸就有更高的要求吧（需要做的更大） 毕竟在pcd文件中能够查找到精确的圆心 是最终标定精度高的重要前提；
我在标定过程中由于场景的限制，没办法同时覆盖图像中的所有位置用于标定相机的内参和畸变，因此我把代码中优化相机内参和畸变的功能关闭了，使用的是原有的参数；另外由于我使用的lidar是32线的 因此我增大了标定板的尺寸，确保每个圆孔中都有3到4条线穿过；
对于实际标定  数据采集的过程中 对数据有哪些确切的要求吗？ 比如说标定板是否要求一直和相机的Z轴垂直，标定板未完全正对相机的时候需要垂直Z轴吗？或者说不标定内参的话还需要采集左前方或者右前方的数据吗？标定板的平面是否允许小范围内的弯曲？
非常感谢您的答复。

wuyuetao, 2024-03-16:
请问你是按照这篇论文给的标定板参数制作的标定板吗？
[image]

我看他们论文给的标定板中棋盘格区域的棋盘格不是正方形
还是没按照他们论文给的标定板给的自己制作的标定板？