PJLab-ADG/SensorsCalibration #57: Target-based lidar2camer - Size of check board pattern for calibration board

TheExplorer95, 2023-02-23:
Hey guys,
in Figure 7 of your [paper](https://arxiv.org/pdf/2205.14087.pdf), you show the calibration board used for the target-based lidar2camera calibration method. I'm curious about the exact sizing of the checkboard patterns, as I couldn't find any references.
Based on your drawing, it is 900mm x 390mm large with 18x8 squares. Correct me if I'm wrong, but this results in 50x48.75 mm sized squares, which seems odd.

The calibration board:
[image]

Some guidance is highly appreciated :)

Best Janosch

xiaokn, 2023-02-24:
> 
Sorry, you are right, there is an error in the picture we put in this paper, and we will make a correction later.

bluecamel, 2023-05-17:
What are the correct dimensions?  Is it that the circles have a diameter of 200mm?

TheExplorer95, 2023-05-18:
Unfortunately, I have no clue so far and am still waiting for the correct values. I've changed to [ATOM](https://github.com/lardemua/atom) for now, which works fine, but does not provide fully automatized labeling of the point cloud data.