PJLab-ADG/SensorsCalibration #58: 手动标定、自动标定、在线标定的区别是什么

zhaoxuesong13, 2023-03-11:
1、三者区别的核心概念是什么？
2、尤其是自动标定转成在线标定在数据输入、参数设置、其他配置和适应性调整有哪些需要注意？
3、lidar2imu中，手动和自动标定过程中，lidar与imu外参误差与lidar可视化中的线特征的粗细（锐利程度）之间是怎么产生关联的？
感谢小伙伴们指导。

xiaokn, 2023-03-13:
- 手动标定需要手动参与
- 自动标定不需要手动参与，但是有一些环境录制数据的要求
- 在线标定是不需要环境和录制数据的要求，在正常行驶下进行标定
- 一般lidar2惯导标定越精确，说明匹配的比较精确，进而拼接的点云特征会越细

pedestrain123, 2024-09-04:
> * 手动标定需要手动参与
> * 自动标定不需要手动参与，但是有一些环境录制数据的要求
> * 在线标定是不需要环境和录制数据的要求，在正常行驶下进行标定
> * 一般lidar2惯导标定越精确，说明匹配的比较精确，进而拼接的点云特征会越细

可以提供一下，lidar2imu online_calib的数据包吗？

zhaoxuesong13, 2024-09-04:
这是来自QQ邮箱的假期自动回复邮件。来信已收到，我会尽快阅读您的邮件。              赵雪松 TEL:18373151973

pedestrain123, 2024-09-04:
@xiaokn 可以提供一下，lidar2imu online_calib的数据包吗？

pedestrain123, 2024-09-05:
@xiaokn 您好作者，我用您的autocalib中公开的数据进行测试，我以gnss坐标系为car的中心坐标系，为什么在线手眼标定出来的结果误差特别大啊？