PJLab-ADG/SensorsCalibration #60: camera-imu标定中CalibrationExPosition()函数

smilebetterworld, 2023-03-15:
camera-imu标定中CalibrationExPosition()函数，这个是在vins基础上增加的吗，请问下背后理论是什么，有参考资料吗，谢谢

jiahao68, 2023-03-16:
> camera-imu标定中CalibrationExPosition()函数，这个是在vins基础上增加的吗，请问下背后理论是什么，有参考资料吗，谢谢

是在vins基础上加的，vins只是对旋转进行了标定，没有对平移进行标定。该部分运用的是手眼标定的方法，相关资料可以在网上搜索到

smilebetterworld, 2023-03-16:
谢谢反馈，这函数里面的核函数是不是没起到作用？
[image]

jiahao68, 2023-03-17:
起到作用了，当distance大于10之后，值越大，其权重越小

> 谢谢反馈，这函数里面的核函数是不是没起到作用？ [image]

起到作用了，当distance大于10之后，值越大，其权重越小

jiahao68, 2023-03-17:
> 谢谢反馈，这函数里面的核函数是不是没起到作用？ [image]

起到作用了，当distance大于10之后，值越大，其权重越小

smilebetterworld, 2023-03-22:
但是A1和b1都乘以huber，不是可以消掉吗，相当于没乘，这样huber并没有起到作用