PJLab-ADG/SensorsCalibration #66: joint_Calib中vec_points_cir坐标的问题

hugestar007, 2023-04-17:
请问大佬，joint_Calib中的vec_points_cir值(1.82f, -3.12f, 0.0f)，(14.18f, -3.12f, 0.0f)，(1.82f, 9.12f, 0.0f)，(14.18f, 9.12f, 0.0f)是carla场景下根据camera的fov来赋的初始值吗？在实际标定中换成camera的fov初始值即可？

xiaokn, 2023-04-18:
> 请问大佬，joint_Calib中的vec_points_cir值(1.82f, -3.12f, 0.0f)，(14.18f, -3.12f, 0.0f)，(1.82f, 9.12f, 0.0f)，(14.18f, 9.12f, 0.0f)是carla场景下标定板的尺寸来赋的初始值吗？标定板的坐标系是怎样定义的呢？在实际标定中换成实际标定板的尺寸即可？

是的，换成你自己的标定板尺寸即可

hugestar007, 2023-04-18:
谢谢大佬的回复。我主要是根据这个坐标(1.82f, -3.12f, 0.0f)，(14.18f, -3.12f, 0.0f)，(1.82f, 9.12f, 0.0f)，(14.18f, 9.12f, 0.0f) 这个标定板原点在棋盘格的左上角定义为标定板的原点。
实际上标定板坐标系自己定义就可以是吗？不管原点在左上角，或者左下角，只需尺寸单位一致就可以？

xiaokn, 2023-04-18:
> 上标定板坐标系自己定义就可以是吗？不管原点在左上角，或者左下角，只需尺寸单位一致就可以？

是的，但是标定板因为是对称的，为了让所有识别到的标定板坐标系保持一致可能需要进行一些处理。

hugestar007, 2023-04-18:
明白，多谢

Elaine211, 2023-08-23:
> 

> 谢谢大佬的回复。我主要是根据这个坐标(1.82f, -3.12f, 0.0f)，(14.18f, -3.12f, 0.0f)，(1.82f, 9.12f, 0.0f)，(14.18f, 9.12f, 0.0f) 这个标定板原点在棋盘格的左上角定义为标定板的原点。 实际上标定板坐标系自己定义就可以是吗？不管原点在左上角，或者左下角，只需尺寸单位一致就可以？

您好，我不太清楚(1.82f, -3.12f, 0.0f)，(14.18f, -3.12f, 0.0f)，(1.82f, 9.12f, 0.0f)，(14.18f, 9.12f, 0.0f) 这几个坐标的数值具体是如何得来的，您能解答一下吗？我看到了论文中标定板的尺寸，我不知道怎么对应这四个坐标。

wangwenzhao21, 2024-05-15:
> > 
> 
> > 谢谢大佬的回复。我主要是根据这个坐标(1.82f, -3.12f, 0.0f)，(14.18f, -3.12f, 0.0f)，(1.82f, 9.12f, 0.0f)，(14.18f) , 9.12f, 0.0f) 这个标定板原点在棋盘格的左上角定义为标定板的原点。另外标定板坐标系自己定义的就是什么？不管原点在左上角，还是左下角，只要尺寸单位一致就可以吗？
> 
> 您好，我不太清楚(1.82f, -3.12f, 0.0f)，(14.18f, -3.12f, 0.0f)，(1.82f, 9.12f, 0.0f)，(14.18f, 9.12f, 0.0 f) 这四个坐标的数值具体是怎么得来的，你能解答一下吗？我看到了论文中标定板的尺寸，我不知道怎么对应这四个坐标。
如何设置棋盘格内点的坐标，圆心坐标就如何设置，注意保持尺度大小一样，比如一个方格边长为5cm对应单位1。测试推理标定板的指定位置（顶点）像素坐标是成功的！
[image]