PJLab-ADG/SensorsCalibration #67: joint_Calib中_imags_pts_cir2D_true的计算问题

hugestar007, 2023-04-19:
[image]
请问大佬，在cv::calibrateCamera标定出相机内外参之后，分为两步，一步是重投影计算_imags_pts_cir2D_true的中心，另外一步是优化参数。为什么先优化参数，然后用优化后的内外参再重投影计算_imags_pts_cir2D_true的中心呢？谢谢

xiaokn, 2023-04-19:
> [image] 请问大佬，在cv::calibrateCamera标定出相机内外参之后，分为两步，一步是重投影计算_imags_pts_cir2D_true的中心，另外一步是优化参数。为什么先优化参数，然后用优化后的内外参再重投影计算_imags_pts_cir2D_true的中心呢？谢谢

这里是两种类型的点 :
(1) 是图像中的圆洞的点，这个是靠相机内参和相机-标定板外参计算出图像域的圆心点
(2) 是雷达三维点在图像上的投影点，这个是靠相机内参和相机-雷达外参计算出雷达点的投影点

hugestar007, 2023-04-19:
我理解是这样：
1.  cv::calibrateCamera(_boards_pts_3d, _imgs_pts, image_size, camera_matrix,
                          k, rvecsMat, tvecsMat, CV_CALIB_FIX_PRINCIPAL_POINT);
       计算出图像序列中每帧的标定板-相机的外参，和相机内参。
2.  根据1的结果，cv::projectPoints(_boards_pts_cir[i], rvecsMat[i], tvecsMat[i],
                      camera_matrix, k, imgpoints_cir);重投影计算图像域的圆心点。
3.  根据1的结果， this->refine_all(camera_intrinsic, distort, vec_extrinsics); 标定板-相机内外参进行两次优化，（区别畸变第一次2个参数，第二次是3个参数）得到优化后的标定板-相机的外参和相机内参。然后和雷达点配准，再优化得到雷达-相机的外参。
我的问题是第2步计算重投影如果用第3步先优化后的标定板-相机的外参和相机内参计算第2步会不会精度更高呢？

hugestar007, 2023-04-25:
@xiaokn

xiaokn, 2023-04-26:
> 我理解是这样：
>
> 1. cv::calibrateCamera(_boards_pts_3d, _imgs_pts, image_size, camera_matrix,
>    k, rvecsMat, tvecsMat, CV_CALIB_FIX_PRINCIPAL_POINT);
>    计算出图像序列中每帧的标定板-相机的外参，和相机内参。
> 2. [...]

最后一步优化是优化雷达和相机的配准，然后加上棋盘格对相机内参以及标定板-相机外参的约束，优化是一起的。所以被成为联合优化

hugestar007, 2023-04-26:
> 最后一步优化是优化雷达和相机的配准，然后加上棋盘格对相机内参以及标定板-相机外参的约束，优化是一起的。所以被成为联合优化

谢谢大神，主要我理解第二步计算重投影现在直接用第一步的结果，理论上先优化下再使用优化后的结果计算重投影精度会略高，当然具体要根据结果来看，所以想问下这步验证您是否用过。谢谢

zouzhiwei217, 2023-07-10:
您好,我也想知道该问题的结果.还有我发现,有时候3D圆心检测算法结果圆心点在地面上,请问您是否设置可更准确的xyz阈值,或者是使用点云算法对标定板进行匹配.