PJLab-ADG/SensorsCalibration #71: 显示问题

WeiJunREN, 2023-05-09:
官方github提供的docker中无法正常显示标定结果可视化界面
报错：terminate called after throwing an instance of 'std::runtime_error'
what(): Pangolin X11: Failed to open X display
Aborted (core dumped)。

docker内export DISPLAY与宿主机一样也无法解决。
请问应该怎么解决

xiaokn, 2023-05-09:
> 官方github提供的docker中无法正常显示标定结果可视化界面 报错：terminate called after throwing an instance of 'std::runtime_error' what(): Pangolin X11: Failed to open X display Aborted (core dumped)。
> 
> docker内export DISPLAY与宿主机一样也无法解决。 请问应该怎么解决

可以参考以下这个issue，看是否能解决 https://github.com/PJLab-ADG/SensorsCalibration/issues/4

WeiJunREN, 2023-05-11:
It helps!
Thanks!

WeiJunREN, 2023-05-11:
It helps!
Thanks!

WeiJunREN, 2023-05-11:
It helps!
Thanks!

Chenwei-Huang, 2023-05-24:
> > 官方github提供的docker中无法正常显示标定结果可视化界面 报错：terminate called after throwing an instance of 'std::runtime_error' what(): Pangolin X11: Failed to open X display Aborted (core dumped)。
> > docker内export DISPLAY与宿主机一样也无法解决。 请问应该怎么解决
> 
> 可以参考以下这个issue，看是否能解决 #4

这个链接好像失效了，里面并没有回答，能不能再解答一下呢

Chenwei-Huang, 2023-05-24:
> > 官方github提供的docker中无法正常显示标定结果可视化界面 报错：terminate called after throwing an instance of 'std::runtime_error' what(): Pangolin X11: Failed to open X display Aborted (core dumped)。
> > docker内export DISPLAY与宿主机一样也无法解决。 请问应该怎么解决
> 
> 可以参考以下这个issue，看是否能解决 #4

问题#4中的那个链接[https://github.com/BayRanger/SensorsCalibration](https://github.com/PJLab-ADG/SensorsCalibration/issues/url)好像失效了，里面没有内容，请问还有别的解决办法吗

iyaqiao, 2023-05-25:
Same question

WeiJunREN, 2023-05-25:
不要直接点进去，复制链接https://github.com/BayRanger/SensorsCalibration，新开网页进再试试呢

WeiJunREN, 2023-05-25:
> 问题#4中的那个链接[https://github.com/BayRanger/SensorsCalibration](https://github.com/PJLab-ADG/SensorsCalibration/issues/url)好像失效了，里面没有内容，请问还有别的解决办法吗

不要直接点进去，复制链接[https://github.com/BayRanger/SensorsCalibration，新开网页进再试试

WeiJunREN, 2023-05-25:
> 不要直接点进去，复制链接[https://github.com/BayRanger/SensorsCalibration，新开网页进再试试

不要直接点进去，复制链接[https://github.com/BayRanger/SensorsCalibration，新开网页进再试试

closed 2023-05-25 by WeiJunREN

Chenwei-Huang, 2023-05-25:
> 不要直接点进去，复制链接[https://github.com/BayRanger/SensorsCalibration，新开网页进再试试呢](https://github.com/BayRanger/SensorsCalibration%EF%BC%8C%E6%96%B0%E5%BC%80%E7%BD%91%E9%A1%B5%E8%BF%9B%E5%86%8D%E8%AF%95%E8%AF%95%E5%91%A2) @Chenwei-Huang @iyaqiao

解决了，感谢~