PJLab-ADG/SensorsCalibration #72: 关于lidar2imu的一点疑问

LittleDang, 2023-05-09:
./bin/run_lidar2imu /share/data/top_center_lidar/ /share/data/top_center_lidar-pose.txt /share/data/gnss-to-top_center_lidar-extrinsic.json

你好，感谢开源了这么棒的工具。
我在使用lidar2imu的时候，遇到了显然不正确的结果。对此，我对所使用到的参数进行的检查，以确保我没法理解错参数的意思。

NovAtel-pose-lidar-time.txt：INS输出的值，需要对齐到每一帧pcd，按照R11 R12 R13 Tx R21 R22 R23 Ty R31 R32 R33 Tz的顺序排列，并且是INS在某个固定的坐标系下的值。
gnss-to-top_center_lidar-extrinsic.json：里面存在的是在INS坐标系下雷达的坐标系，即Tins2lidar，以一个4*4的矩阵形式表示，并且是行主序的。
top_center_lidar：每一帧点云保存为一个pcd，名称为对应的时间戳

运行指令：
./bin/run_lidar2imu data/top_center_lidar/ data/NovAtel-pose-lidar-time.txt data/gnss-to-top_center_lidar-extrinsic.json

最终得到的标定结果似乎与正确值有较大差距

在阅读了源码之后，发现了一个地方令我有点困惑

在[SensorsCalibration/lidar2imu/auto_calib/src
/calibration.cpp](https://github.com/PJLab-ADG/SensorsCalibration/blob/master/lidar2imu/auto_calib/src/calibration.cpp)的第46行
`Ti *= Tl2i;`

是否应该改成

`Ti *= Tl2i.inverse();`

我尝试着修改之后，得到的结果似乎比较接近正确值。

LittleDang, 2023-05-09:
使用官方所提供的数据的运行得到的差异似乎比较小一点，但是修改后delta T的值依然更小，我不太确定哪个是更加正确结果

修改后
```bash
delta T is:
   0.999943  0.00726685  0.00777305  -0.0125842
-0.00721024    0.999947 -0.00728733   0.0646276
 -0.0078256  0.00723087    0.999943           0
          0           0           0           1
refined T(imu 2 lidar):
 0.00760922     0.99997  0.00144469    -1.18388
  -0.999673  0.00757167   0.0244252   -0.105042
  0.0244136 -0.00163008      0.9997    -1.38587
         -0           0          -0           1
```

修改前
```bash
delta T is:
   0.999602   0.0278781 -0.00434395   -0.103252
 -0.0279222    0.999555   -0.010448   0.0364158
 0.00405075   0.0105652    0.999936           0
          0           0           0           1
refined T(imu 2 lidar): 
 0.0285213   0.999509  0.0129608   -1.10819
 -0.999213  0.0281505  0.0279458  -0.103511
 0.0275673 -0.0137477   0.999525    -1.3718
        -0          0         -0          1

```

jiahao68, 2023-05-09:
> 你好，感谢开源了这么棒的工具。 我在使用lidar2imu的时候，遇到了显然不正确的结果。对此，我对所使用到的参数进行的检查，以确保我没法理解错参数的意思。
>
> NovAtel-pose-lidar-time.txt：INS输出的值，需要对齐到每一帧pcd，按照R11 R12 R13 Tx R21 R22 R23 Ty R31 R32 R33 [...]

Ti *= Tl2i得到的是每帧lidar在以第0帧novatel为坐标原点时的pose。
gnss-to-top_center_lidar-extrinsic.json中存放的是novatel坐标系到lidar坐标系的外参，比如，当lidar坐标系是x轴朝前，y轴朝左,z轴朝上；novatel坐标系x轴朝右，y朝前，z朝上时。此时的外参中的旋转矩阵应该是：
0，1，0
-1，0，0
0，0，1
结果明显不正确可能是这个外参弄反了，即使用成了lidar到novatel的外参。你可以给你的外参取逆试试效果

LittleDang, 2023-05-09:
感谢回复，我对json里的值取逆后结果正确了。
应该是我对记号的理解与存在差异，按照您给出的例子我算出来的旋转矩阵是

R^{novatel}_{lidar}
=R_z(90)
=0,-1,0
=1,0,0
=0,0,1

结果与您存在一个逆的差异。

closed 2023-05-10 by LittleDang

pedestrain123, 2024-07-06:
> 感谢回复，我对json里的值取逆后结果正确了。 应该是我对记号的理解与存在差异，按照您给出的例子我算出来的旋转矩阵是
> 
> R^{novatel}_{激光雷达} =R_z(90) =0,-1,0 =1,0,0 =0,0,1
> 
> 结果与您存在一个逆的差异。

你好，兄弟问一下NovAtel-pose-lidar-time.txt存储的应该是ins到初始时刻的位姿变换吧？

lixiuyu-cumt, 2024-07-18:
> > 感谢回复，我对json里的值取逆后结果正确了。 应该是我对记号的理解与存在差异，按照您给出的例子我算出来的旋转矩阵是
> > R^{novatel}_{激光雷达} =R_z(90) =0,-1,0 =1,0,0 =0,0,1
> > 结果与您存在一个逆的差异。
> 
> 你好，兄弟问一下NovAtel-pose-lidar-time.txt存储的应该是ins到初始时刻的位姿变换吧？

哈喽，请问你现在搞清楚这一块了吗

EItByTe, 2024-08-06:
> > > 感谢回复，我对json里的值取逆后结果正确了。 应该是我对记号的理解与存在差异，按照您给出的例子我算出来的旋转矩阵是
> > > R^{novatel}_{激光雷达} =R_z(90) =0,-1,0 =1,0,0 =0,0,1
> > > 结果与您存在一个逆的差异。
> > 
> > 
> > 你好，兄弟问一下NovAtel-pose-lidar-time.txt存储的应该是ins到初始时刻的位姿变换吧？
> 
> 哈喽，请问你现在搞清楚这一块了吗

hi，请问您现在搞清楚这一块了吗？