PJLab-ADG/SensorsCalibration #73: SensorX2Car  calib yaw  with Segmentation fault

lyw615, 2023-05-12:
Environment:  xiaokyan/opencalib:v1  and      scllovewkf/opencalib:v1

Following the readMe to operate ,  and  the program  calculates  roll and pitch  successfully.   The program  break  down  at the line  `BSpline bspline_x = BSpline::Builder(sample_x).degree(bspine_degree_).smoothing(BSpline::Smoothing::PSPLINE).alpha(0.03).build();` of  function GetYawSegs.  The debug information  as  followings. Any  reply  is appreciated.

` BSpline bspline_x = BSpline::Builder(sample_x).degree(bspine_degree_).smoothing(BSpline::Smoothing::PSPLINE).alpha(0.03).build();
(gdb) n

Thread 1 "run_lidar2car" received signal SIGSEGV, Segmentation fault.
0x0000000001c298b9 in Eigen::SparseCompressedBase<Eigen::SparseMatrix<double, 0, int> >::InnerIterator::InnerIterator (this=0x7fff49359340, mat=..., outer=0)
    at /usr/include/eigen3/Eigen/src/SparseCore/SparseCompressedBase.h:120
120	        m_id = mat.outerIndexPtr()[outer];
`

sisyphus-99, 2023-05-12:
It may be because the integrated SLAM algorithm(FAST-LOAM) doesn't run well. Could you see the trajectory image in the output folder to check if the x-y trajectory is right ?

lyw615, 2023-05-12:
> It may be because the integrated SLAM algorithm(FAST-LOAM) doesn't run well. Could you see the trajectory image in the output folder to check if the x-y trajectory is right ?

The program break down with the example data  offered in readme.  I  have checked the  x-y trajectory is  right.  It's  same as  trajectory  in demo picture.

sisyphus-99, 2023-05-12:
Sorry but we haven't met this error before. It looks to be related to the SPLINTER library. 
Here is an example provided by the library: https://github.com/bgrimstad/splinter/blob/master/docs/cpp_interface.md 
Maybe you can test it first to see if the library works well.

lyw615, 2023-05-16:
Before  compiling run_lidar2car,  SPLINTER   should be compiled and set .so file to  lib  directory.

sisyphus-99, 2023-05-16:
Yes, splinter is used as a static library and .a file is put in the lib directory. I suggest checking the library because the example data has been tested successfully in the same docker enviroment. Since your trajectory output is right, I suppose nothing wrong with `sample_x`(build with lidar pose x)  in the error line: `BSpline::Builder(sample_x).degree(bspine_degree_).smoothing(BSpline::Smoothing::PSPLINE).alpha(0.03).build();`
Then maybe there is problem with the library. To test it, you can just replace the main function in `app/run_lidar2car.cpp` by the example usage of splinter.

pangkaka2010, 2023-05-18:
> Before compiling run_lidar2car, SPLINTER should be compiled and set .so file to lib directory.

I also encountered the same problem, how did you solve it at that time?

sisyphus-99, 2023-05-19:
> > Before compiling run_lidar2car, SPLINTER should be compiled and set .so file to lib directory.
> 
> I also encountered the same problem, how did you solve it at that time?

The lib file `libsplinter-static-3-0.a` was now uploaded. See [here](https://github.com/PJLab-ADG/SensorsCalibration/tree/master/SensorX2car/lidar2car/lib). Putting it under `lib` directory should solve this problem.

closed 2023-05-29 by lyw615