PJLab-ADG/SensorsCalibration #75: Stability checking of VP's position in camera2car auto calibration algorithm

SternSu, 2023-05-17:
Hi, I have a question for camera2car auto calibration algorithm. I found that there are following words in the paper _SensorX2car: Sensors-to-car calibration for autonomous driving in road scenarios_: Note that the vehicle doesn’t need to travel straightly all the time. We use the stability of the position of the vanishing point as a judgment condition in our algorithm to extract useful segments. 

But when I used image sequences collected in a trajectory composed of both straight part and curve part, the calibration results were pretty bad. I use the pretrained model and it works well for images representing straight motion, so I think the problem is not in the model.
 
I am wondering if there are codes conducting the stability checking of VP's position as the paper says? When I was checking the code, it seems that there are no such kind of things.

Thank you very much for your help!

xiaokn, 2023-05-17:
> 
You can add the code, which is very simple. For example, when it is judged that the vanishing point changes very little, it is considered that the vehicle is walking in a straight line, and only this part is used for calibration.

SternSu, 2023-05-21:
> > 
> 
> You can add the code, which is very simple. For example, when it is judged that the vanishing point changes very little, it is considered that the vehicle is walking in a straight line, and only this part is used for calibration.

Thank you for your answers, I will try to add the relevant code.