PJLab-ADG/SensorsCalibration #78: 代码细节问题

WeiJunREN, 2023-05-23:
lidar2camera/auto_calib/src/optimizer.cpp：49：76
for (int i1 = -5; i1 < 5; i1++) {
      for (int i2 = -5; i2 < 5; i2++) {
        for (int i3 = -5; i3 < 5; i3++) {
          for (int i4 = -5; i4 < 5; i4++) {
            for (int i5 = -5; i5 < 5; i5++) {
              for (int i6 = -5; i6 < 5; i6++) {
                var[0] = i1 * rpy_resolution;
                var[1] = i2 * rpy_resolution;
                var[2] = i3 * rpy_resolution;
                var[3] = i4 * xyz_resolution;
                var[4] = i5 * xyz_resolution;
                var[5] = i6 * xyz_resolution;
                float cnt = CostFunction(var, distance_img, register_cloud);
                if (cnt > maxPointCnt) {
                  maxPointCnt = cnt;
                  for (size_t k = 0; k < 6; k++) {
                    bestVal[k] = var[k];
                  }

                  std::cout << "match point increase to: " << maxPointCnt
                            << std::endl;
                }
              }
            }
          }
        }
      }
咨询一下，这里循环的作用是什么呢？加上if成立一共7层嵌套循环，必要性是？不会影响代码执行效率吗

xiaokn, 2023-05-23:
> lidar2camera/auto_calib/src/optimizer.cpp：49：76 for (int i1 = -5; i1 < 5; i1++) { for (int i2 = -5; i2 < 5; i2++) { for (int i3 = -5; i3 < 5; i3++) { for (int i4 = -5; i4 < 5; i4++) { for (int i5 = [...]

这里实际应该没有用到，只是起到一个说明或者注释的作用

WeiJunREN, 2023-05-25:
好的，他实际是暴力搜索，但是个人感觉效率有点低，试着问问能不能优化一下

xiaokn, 2023-05-25:
> 好的，他实际是暴力搜索，但是个人感觉效率有点低，试着问问能不能优化一下

他实际不是暴力搜索，这段代码被bool is_violence_search = false; if (is_violence_search) 注释掉了，也就是删除也不影响

WeiJunREN, 2023-05-25:
我知道是被注释了，可能是我刚刚没表达清楚，就是这一串代码表达的应该是通过暴力搜索的方法查找xyzrpy吧，但是我觉得暴力搜索效率不太高，这一段能不能优化一下呢

xiaokn, 2023-05-25:
> 我知道是被注释了，可能是我刚刚没表达清楚，就是这一串代码表达的应该是通过暴力搜索的方法查找xyzrpy吧，但是我觉得暴力搜索效率不太高，这一段能不能优化一下呢

嗯嗯，应该是可以优化的，这里可以尝试采用优化的方法解决，你可以探索一下。

WeiJunREN, 2023-05-25:
哈哈，好的发自我的手机-------- 原始邮件 --------发件人： xiaokn ***@***.***>日期： 2023年5月25日周四 下午2:04收件人： PJLab-ADG/SensorsCalibration ***@***.***>抄送： WeiJunREN ***@***.***>, Author ***@***.***>主    题： Re: [PJLab-ADG/SensorsCalibration] 代码细节问题 (Issue #78)

我知道是被注释了，可能是我刚刚没表达清楚，就是这一串代码表达的应该是通过暴力搜索的方法查找xyzrpy吧，但是我觉得暴力搜索效率不太高，这一段能不能优化一下呢

嗯嗯，应该是可以优化的，这里可以尝试采用优化的方法解决，你可以探索一下。

Muxindawang, 2023-05-30:
> 他实际是暴力搜索，但是个人感觉效率有点低，试着问问能不能优化一下

哈喽 我最近也在使用这个auto标定的工程  想咨询下 你那边使用这个工程的结果怎么样？大概能标定多少度以内的误差。我在使用过程中，将yaw角增加了5度之后，发现并没有匹配上车道线和杆；下面图中的0.05我理解应该是修正矩阵的欧拉角的范围（-0.05~0.05度），如果我理解的正确的话，那么修正的幅度太有限了，在实际使用过程中几乎起不到什么作用。期待您的回复，谢谢。
[image]

closed 2023-06-01 by WeiJunREN