PJLab-ADG/SensorsCalibration #79: 关于lidar2imu相关的输入数据采集问题？

maomao0220, 2023-05-26:
您好，想咨询您几个关于lidar2imu数据的问题，下面两个问题均在同步的前提下？希望得到您的回复
1、lidar_pcds_dir：是存储在自身坐标系下的lidar原始数据吗？
2、lidar_pose_file：存储的数据是直接保存自身坐标系下的惯导数据吗？需不需要其他的处理
期待您的回复

xiaokn, 2023-05-26:
> 您好，想咨询您几个关于lidar2imu数据的问题，下面两个问题均在同步的前提下？希望得到您的回复 1、lidar_pcds_dir：是存储在自身坐标系下的lidar原始数据吗？ 2、lidar_pose_file：存储的数据是直接保存自身坐标系下的惯导数据吗？需不需要其他的处理 期待您的回复

1. 是的，原始的PCD文件
2. 是的，惯导的位姿。需要注意的是这里只优化雷达到惯导的外参，如果需要标定imu内参的话，需要修改代码加上imu内参标定部分

maomao0220, 2023-05-31:
@xiaokn 您好，非常抱歉再次打扰您，关于lidar pcd的存储，我想问一下：
在生成pcd过程中，是每隔一段时间存储当前时刻的一帧pcd还是实时存储每一帧的数据又或者是存储累积一段时间的pcd？期待您的回复

jiahao68, 2023-06-05:
> 

实时存储每一帧的数据即可

jiahao68, 2023-06-05:
> @xiaokn 您好，非常抱歉再次打扰您，关于lidar pcd的存储，我想问一下： 在生成pcd过程中，是每隔一段时间存储当前时刻的一帧pcd还是实时存储每一帧的数据又或者是存储累积一段时间的pcd？期待您的回复

实时存储每一帧的数据即可

Dlemand, 2023-06-05:
hello . I find there are 12 data in imu data for each line. is it the pose data without [0,0,0,1]-the last line ?  but how can i get the 12 data from imu .

w-shuw, 2023-07-06:
您好，请教一下lidar_pose_file的数据是直接imu话题数据转旋转矩阵得到的吗