PJLab-ADG/SensorsCalibration #82: lidar2camera中auto部分代码问题

Muxindawang, 2023-06-06:
您好，我在使用auto lidar2camera部分代码时，发现一个问题，var数组中的前三位对应的到底谁roll pitch yaw 还是 x y z？
[image]
在这个GetDetalT中显然 前三位对应的时旋转角；
[image]
但是在random search params中 前三位的搜索范围使用的是xyz的随机变量，虽然这并不影响功能，但这么写是会引起歧义的。
并且上面的传参中，三个值的范围是-0.05~0.05这个对应的是角度而不是弧度，那么这个修正范围是不是太小了
[image]
希望能收到您的回复 谢谢

xiaokn, 2023-06-06:
> 您好，我在使用auto lidar2camera部分代码时，发现一个问题，var数组中的前三位对应的到底谁roll pitch yaw 还是 x y z？ [image] 在这个GetDetalT中显然 前三位对应的时旋转角； [image] 但是在random search params中 前三位的搜索范围使用的是xyz的随机变量，虽然这并不影响功能，但这么写是会引起歧义的。 并且上面的传参中，三个值的范围是-0.05~0.05这个对应的是角度而不是弧度，那么这个修正范围是不是太小了 [image] 希望能收到您的回复 谢谢

随机搜索的方法，是从粗到细的搜索方法。刚开始大范围搜索，最后在小范围搜索

xiaokn, 2023-06-06:
> 您好，我在使用auto lidar2camera部分代码时，发现一个问题，var数组中的前三位对应的到底谁roll pitch yaw 还是 x y z？ [image] 在这个GetDetalT中显然 前三位对应的时旋转角； [image] 但是在random search params中 前三位的搜索范围使用的是xyz的随机变量，虽然这并不影响功能，但这么写是会引起歧义的。 并且上面的传参中，三个值的范围是-0.05~0.05这个对应的是角度而不是弧度，那么这个修正范围是不是太小了 [image] 希望能收到您的回复 谢谢

这个范围可以自行去设定，设置多大根据自己的需求来

Muxindawang, 2023-06-06:
> 随机搜索的方法，是从粗到细的搜索方法。刚开始大范围搜索，最后在小范围搜索

是的   另外一个问题 就是有歧义的那部分 个人认为还是有必要修改一下

Muxindawang, 2023-06-06:
> > 随机搜索的方法，是从粗到细的搜索方法。刚开始大范围搜索，最后在小范围搜索
> 
> 是的 另外一个问题 就是有歧义的那部分 个人认为还是有必要修改一下

还想请教一下  您有测试过优化的范围吗  比如说pitch角在变化多少度以内  这个方法能够得到一个比较准确的值？ 然后每次搜索的范围 根据经验来谈 应该怎么设置呢，上一次搜索的范围和下一次搜索的范围之间的值是10倍或者5倍或者2倍，这个能传授下经验吗

Muxindawang, 2023-06-06:
> > > 随机搜索的方法，是从粗到细的搜索方法。刚开始大范围搜索，最后在小范围搜索
> > 
> > 
> > 是的 另外一个问题 就是有歧义的那部分 个人认为还是有必要修改一下
> 
> 感谢您的建议，后面有时间了我们会重构一下OpenCalib的代码

非常期待您的重构  还有论文中P3L的 内容貌似并没有在代码中有所体现  这部分 有考虑添加吗

xiaokn, 2023-06-06:
> > > 随机搜索的方法，是从粗到细的搜索方法。刚开始大范围搜索，最后在小范围搜索
> > 
> > 
> > 是的 另外一个问题 就是有歧义的那部分 个人认为还是有必要修改一下
> 
> 还想请教一下 您有测试过优化的范围吗 比如说pitch角在变化多少度以内 这个方法能够得到一个比较准确的值？ 然后每次搜索的范围 根据经验来谈 应该怎么设置呢，上一次搜索的范围和下一次搜索的范围之间的值是10倍或者5倍或者2倍，这个能传授下经验吗

比如，你粗标定的步长是1，那么下一次循环的范围理论上就可以设置为1，步长0.1，下一次范围为0.1，步长为。。。但是实际可能比这个理论范围设置的大一点

xiaokn, 2023-06-06:
> > > > 随机搜索的方法，是从粗到细的搜索方法。刚开始大范围搜索，最后在小范围搜索
> > > 
> > > 
> > > 是的 另外一个问题 就是有歧义的那部分 个人认为还是有必要修改一下
> > 
> > 
> > 感谢您的建议，后面有时间了我们会重构一下OpenCalib的代码
> 
> 非常期待您的重构 还有论文中P3L的 内容貌似并没有在代码中有所体现 这部分 有考虑添加吗

P3L很简单的，根据正交关系求一个初始外参，可以自己加一下，因为一些原因，我们删除了这部分代码。
另外这个方法要求条件还是比较苛刻的，可以关注一下我们最新的工作https://arxiv.org/abs/2306.02656

Muxindawang, 2023-06-06:
> > > > 随机搜索的方法，是从粗到细的搜索方法。刚开始大范围搜索，最后在小范围搜索
> > > 
> > > 
> > > 是的 另外一个问题 就是有歧义的那部分 个人认为还是有必要修改一下
> > 
> > 
> > 还想请教一下 您有测试过优化的范围吗 比如说pitch角在变化多少度以内 这个方法能够得到一个比较准确的值？ 然后每次搜索的范围 根据经验来谈 应该怎么设置呢，上一次搜索的范围和下一次搜索的范围之间的值是10倍或者5倍或者2倍，这个能传授下经验吗
> 
> 比如，你粗标定的步长是1，那么下一次循环的范围理论上就可以设置为1，步长0.1，下一次范围为0.1，步长为。。。但是实际可能比这个理论范围设置的大一点

所以比如说 我想修正5度以内的误差  我的第一轮的步长应该设置为5，下一次设置为0.5  在下一次0.05   是这样么  或者我想更精确一些 可以每次递减的更少  依次设置为 5 4 3 2  1 0.5 这样

xiaokn, 2023-06-06:
> 所以比如说 我想修正5度以内的误差 我的第一轮的步长应该设置为5，下一次设置为0.5 在下一次0.05 是这样么 或者我想更精确一些 可以每次递减的更少 依次设置为 5 4 3 2 1 0.5 这样

嗯嗯，还是要根据实际应用去测试一下效果

Muxindawang, 2023-06-06:
> 嗯嗯，还是要根据实际应用去测试一下效果

好的  非常感谢您的回复 期待你们新的研究成果

Muxindawang, 2023-06-06:
> > > > > 随机搜索的方法，是从粗到细的搜索方法。刚开始大范围搜索，最后在小范围搜索
> > > > 
> > > > 
> > > > 是的 另外一个问题 就是有歧义的那部分 个人认为还是有必要修改一下
> > > 
> > > 
> > > 感谢您的建议，后面有时间了我们会重构一下OpenCalib的代码
> > 
> > 
> > 非常期待您的重构 还有论文中P3L的 内容貌似并没有在代码中有所体现 这部分 有考虑添加吗
> 
> P3L很简单的，根据正交关系求一个初始外参，可以自己加一下，因为一些原因，我们删除了这部分代码。 另外这个方法要求条件还是比较苛刻的，可以关注一下我们最新的工作https://arxiv.org/abs/2306.02656

代码什么时候会开源

weihongwei-zg, 2023-12-27:
> 您好，我在使用auto lidar2camera部分代码时，发现一个问题，var数组中的前三位对应的到底谁roll pitch ya

您好，我也正在尝试使用这套标定工具，但是发现自动标定效果很差，且手动标定工具调整参数十分困难（基本上调对一个图像换一个图像又偏差很大），方便指点一下嘛