PJLab-ADG/SensorsCalibration #84: ubuntu22.04 vs ubuntu20.04 libjsoncpp-dev installation

ckinsung85, 2023-06-13:
I git clone the OpenCalib git to two OS ubuntu20.04 and ubuntu22.04

I noticed under ubuntu22.04, I was stucked in the installation of 

sudo apt-get install libjsoncpp-dev

...
[ 50%] Building CXX object CMakeFiles/CameraCalibration.dir/src/run_intrinsic_calibration.cpp.o
[ 66%] Linking CXX static library ../lib/libCameraCalibration.a
[ 66%] Built target CameraCalibration
[ 83%] Building CXX object CMakeFiles/run_intrinsic_calibration.dir/src/run_intrinsic_calibration.cpp.o
[100%] Linking CXX executable ../bin/run_intrinsic_calibration
/usr/bin/ld: cannot find -ljsoncpp: No such file or directory
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/run_intrinsic_calibration.dir/build.make:150: ../bin/run_intrinsic_calibration] Error 1
make[1]: *** [CMakeFiles/Makefile2:111: CMakeFiles/run_intrinsic_calibration.dir/all] Error 2
make: *** [Makefile:91: all] Error 2

Running of 
# mkdir build
mkdir -p build && cd build
# build
cmake .. && make

was not successfull!!!

However, getting stucked for several times, I chose to reinstall my OS from ubuntu22.04 to ubuntu20.04 and the problem was solved.

ckinsung85, 2023-06-13:
Write down the relevant packages installed for the succesful running of the codes

1. Installation of Ubuntu20.04

2. ```
   sudo apt install git
   ```

3. ```
   git clone https://github.com/PJLab-ADG/SensorsCalibration.git
   ```

4. ```
   sudo apt install cmake
   ```

5. ```
   sudo apt-get install build-essential
   sudo apt-get install g++
   ```

6. ```
   sudo apt install libgtk2.0-dev pkg-config
   ```

7. ```
   sudo apt install libcanberra-gtk-module libcanberra-gtk3-module -y
   ```

8. Install OpenCV

   ```
   # Install minimal prerequisites (Ubuntu 18.04 as reference)
   sudo apt update && sudo apt install -y cmake g++ wget unzip
   
   # Download and unpack sources
   wget -O opencv.zip https://github.com/opencv/opencv/archive/4.x.zip
   wget -O opencv_contrib.zip https://github.com/opencv/opencv_contrib/archive/4.x.zip
   unzip opencv.zip
   unzip opencv_contrib.zip
   
   # Create build directory and switch into it
   mkdir -p build && cd build
   
   # Configure
   cmake -DOPENCV_EXTRA_MODULES_PATH=../opencv_contrib-4.x/modules ../opencv-4.x
   
   # Build
   cmake --build .
   
   # 
   sudo make install
   ```

9. ```
   sudo apt install libpcl-dev
   ```

10. ```
    cd ./intrinsic_calib/
    ./bin/run_intrinsic_calibration ./data/
    ```

    ```
    ckinsung@CKS-G3-3590:~/Documents/SensorsCalibration-master/camera_intrinsic/intrinsic_calib$ ./bin/run_intrinsic_calibration ./data/
    image path: ./data/
    Select image 5.png
    Select image 3.png
    Select image 22.png
    Select image 13.png
    Select image 23.png
    Select image 6.png
    Select image 11.png
    Select image 2.png
    Select image 12.png
    Select image 17.png
    Select image 20.png
    Select image 15.png
    Select image 16.png
    Select image 10.png
    Select image 18.png
    Select image 4.png
    Select image 7.png
    Select image 8.png
    Select image 9.png
    Select image 14.png
    Select image 19.png
    Select image 21.png
    input 22 images
    detected 22 images
    
    camera intrinsic: 
    1057.55	0	963.605
    0	1059.61	581.099
    0	0	1
    
    dist: 
    -0.148233, 0.095432, -0.000553691, -0.000477981, -0.0229052, 
    ```

behretv, 2023-09-13:
If you replace  `libjsoncpp.a` with `${JSONCPP_LIBRARIES}` in the CMakeLists.txt file and install cppjson via 
```bash
sudo apt-get install libjsoncpp-dev
```
it should compile without errors.