PJLab-ADG/SensorsCalibration #87: AutoCalibFishEye 的 Ground 坐标系问题

FreemanHsu, 2023-06-15:
Sample 中，前后左右四个鱼眼摄像头的外参如下，

    T_FG << 9.99277118e-01, 3.82390286e-04, -3.80143958e-02, 6.75437418e-01,
        -2.30748265e-02, -7.88582447e-01, -6.14495953e-01, 2.50896883e+01,
        -3.02124625e-02, 6.14928921e-01, -7.88003572e-01, 3.17779305e+00, 0, 0,
        0, 1;
    T_LG << -1.21898860e-02, 9.99924056e-01, -1.81349393e-03, 1.36392943e+00,
        8.02363600e-01, 8.69913885e-03, -5.96772133e-01, 1.60942881e+01,
        -5.96711036e-01, -8.72966581e-03, -8.02408707e-01, 1.04105913e+01, 0, 0,
        0, 1;
    T_BG << -9.99615699e-01, 1.56439861e-02, -2.28849354e-02, 1.09266953e+00,
        2.59906371e-02, 8.16008735e-01, -5.77454960e-01, 2.46308124e+01,
        9.64060983e-03, -5.77827838e-01, -8.16101739e-01, 6.60957845e+00, 0, 0,
        0, 1;
    T_RG << 4.57647596e-03, -9.99989102e-01, 9.22798184e-04, -1.66115120e-01,
        -6.26343448e-01, -3.58584197e-03, -7.79538984e-01, 1.76226207e+01,
        7.79533797e-01, 2.98955282e-03, -6.26353033e-01, 6.08338205e+00, 0, 0,
        0, 1;

我理解这四组外参分别是 Ground 坐标系下，四个 Camera 的位姿
从平移部分来看，数值上还是比较奇怪的，比如 y 分量，分别是 25 16 24 17 量级
请问 Ground 坐标系是如何定义的？这组外参是如何获得的？
谢谢

MobiusLeo, 2023-06-16:
这个参数来源于别的数据集（他将t缩放变大的同时缩放变小BEV相机内参和高度，所以不合逻辑的同时可以得到一致性的结果），具体而言：这里的平移向量是乘以10之后的。你可以将其除以10，这个时候你把BEV内参调整为和环视相机外参差不多大的参数（我设置的是fx，fy=421）再把BEV相机高度设置为6m，得到的结果就可以几乎和之前相当了，这时候t是符合逻辑的。

FreemanHsu, 2023-06-16:
好的我尝试一下

jike5, 2023-12-13:
他这里的内参其中fy是负数，我发现改成正数会导致左右图像上下filp，这里的负数设计是为什么呢

zhagao84, 2024-11-12:
想问下 这个地面坐标的原点位置在哪 尤其是平移分量。我用自己的相机内参怎么调整