PJLab-ADG/SensorsCalibration #88: lidar2imu关于固态雷达和imu数据的问题咨询

huangwefeng121, 2023-06-16:
首先感谢作者开源了这么棒的工具，我是个小白，关于lidar2imu部分有一下疑惑：
1、我使用的是速腾M1雷达，视野范围是正前方正负60度，俯仰是正负25度的范围，是否适用于您的工具箱，因为雷达运动的时候有目标会离开视野范围，会不会导致前后帧配准过程的误差？
2、imu或者是ins需要提供哪些数据信息，我看data文件夹中有四个.txt文本，分别对应什么数据呢？
希望没有打搅您，也期待您的回复

huangwefeng121, 2023-06-18:
hello作者，能帮忙解惑一下吗 @friskit-china @YueDayu @yikang-li @BOBrown

huangwefeng121, 2023-06-24:
作者你好，关于top_center_lidar_pose.txt文件，我想知道里面的旋转矩阵是x、y、z三轴通过什么旋转顺序得到的，旋转过程中是内旋还是外旋，因为我的imu只能得到变化过程的欧拉角，需要根据欧拉角去计算旋转矩阵，现在面临旋转顺序和内旋外旋问题。

jiahao68, 2023-06-25:
> 首先感谢作者开源了这么棒的工具，我是个小白，关于lidar2imu部分有一下疑惑： 1、我使用的是速腾M1雷达，视野范围是正前方正负60度，俯仰是正负25度的范围，是否适用于您的工具箱，因为雷达运动的时候有目标会离开视野范围，会不会导致前后帧配准过程的误差？ 2、imu或者是ins需要提供哪些数据信息，我看data文件夹中有四个.txt文本，分别对应什么数据呢？ 希望没有打搅您，也期待您的回复

1.我们没有实验过固态雷达到惯导的标定。但理论上，只要平面特征充足应该是可以标的，至于雷达运动的时候有目标会离开视野范围这个问题，可能需要你自己修改一下代码，添加一些约束条件
2.ins提供和lidar时间戳对齐后的pose信息，pose中包含3x3的旋转矩阵和3x1的平移向量

jiahao68, 2023-06-25:
> 作者你好，关于top_center_lidar_pose.txt文件，我想知道里面的旋转矩阵是x、y、z三轴通过什么旋转顺序得到的，旋转过程中是内旋还是外旋，因为我的imu只能得到变化过程的欧拉角，需要根据欧拉角去计算旋转矩阵，现在面临旋转顺序和内旋外旋问题。
这个应该得看你自己对roll，pitch和yaw角旋转顺序是怎么定义的或者是你的imu是怎么定义的，旋转矩阵是没有歧义的，有歧义的是你的欧拉角的旋转顺序

huangwefeng121, 2023-06-26:
2.ins提供和lidar时间戳对齐后的pose信息，pose中包含3x3的旋转矩阵和3x1的平移向量

感谢解答，您这边测试时平移向量是通过imu三轴加速度二次积分得到的吗？还是经过了gps或者里程计修正后给出的，我这边通过imu加速度积分得到的平移量误差特别大，标定出来的结果（imu 2 lidar）也是平移量达到了1km。

jiahao68, 2023-06-27:
> 

经过gps修正的，同时我们用的惯导也是精度比较高的。当车速小于10km/h，点云频率>=10Hz时，运动畸变比较小，可以不补偿运动畸变

huangwefeng121, 2023-06-27:
> > 
> 
> 经过gps修正的，同时我们用的惯导也是精度比较高的。当车速小于10km/h，点云频率>=10Hz时，运动畸变比较小，可以不补偿运动畸变

好的，谢谢作者解惑，我这边目前只是用纯imu，后面考虑把gps加进去修正。后面有问题再来向作者请教。

WadeFrank, 2023-08-04:
@huangwefeng121 你好，我最近也在做IMU和固态激光雷达的标定，请问你使用这个代码标定成功了吗？有机会的话可以交流一下

pedestrain123, 2024-07-27:
> @huangwefeng121你好，我最近在做IMU和固态激光雷达的标定，请问你用这个代码标定成功了吗？有机会的话可以交流一下

你好兄弟，请问这个标定工具箱可以标定固态雷达和imu 吗？

pedestrain123, 2024-07-27:
> > > 
> > 
> > 
> > 经过gps修正的，同时我们用的惯导也是精度比较高的。当车速小于10km/h，点云频率>=10Hz时，运动畸变可以变得比较小，不能补偿运动畸变
> 
> 好的，谢谢作者解惑，我目前只是用纯imu，后面考虑把gps加进去修改。后面有问题请向作者指教。

你好兄弟，请问这个标定工具箱可以标定固态雷达和imu 吗？

OKAIOK, 2025-11-10:
> > > > 
> > > 
> > > 
> > > 经过gps修正的，同时我们用的惯导也是精度比较高的。当车速小于10km/h，点云频率>=10Hz时，运动畸变可以变得比较小，不能补偿运动畸变
> > 
> > 
> > 好的，谢谢作者解惑，我目前只是用纯imu，后面考虑把gps加进去修改。后面有问题请向作者指教。
> 
> 你好兄弟，请问这个标定工具箱可以标定固态雷达和imu 吗？

你好哥们，最近我想也用该工具标定固态雷达与rtk，但是好像不适配，请问你这边有成功标定过吗，因为我看源码中，可能需要雷达的ring值，固态雷达一般没有这个值，无法进行优化标定。但是我又不太清除如何根据固态雷达修改源码，能否交流一下？ @jiahao68 @pedestrain123 @wadefrank