PJLab-ADG/SensorsCalibration #91: Docker Versions incorrect for Lidar2Camera

alre5639, 2023-06-23:
The docker container versions are not compatible out of the box with the lidar2camera calibration. Do we need to install the correct versions of the dependencies inside the docker container?

```
cmake .. && make
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at CMakeLists.txt:10 (find_package):
  Could not find a configuration file for package "PCL" that exactly matches
  requested version "1.9".

  The following configuration files were considered but not accepted:

    /usr/local/pcl-1.9/share/pcl-1.9/PCLConfig.cmake, version: 1.9.1
    /usr/lib/x86_64-linux-gnu/cmake/pcl/PCLConfig.cmake, version: 1.7.2
```

xiaokn, 2023-07-11:
> 
I don't have this problem here, and you can install the correct versions?

JiahaoXia, 2023-07-26:
Same error for me. Have you fixed this issue? @alre5639

963070827, 2024-01-07:
> > 
> 
> I don't have this problem here, and you can install the correct versions?

I also have this problem here ,  how can i solve it