PJLab-ADG/SensorsCalibration #96: lidar2car编译问题，/usr/bin/ld: cannot find -lsplinter-static-3-0

yesheng1996, 2023-07-10:
这个splinter-static-3-0是什么模块？要如何安装？每次都报错：
/usr/bin/ld: cannot find -lsplinter-static-3-0
collect2: error: ld returned 1 exit status
CMakeFiles/run_lidar2car.dir/build.make:391: recipe for target '../bin/run_lidar2car' failed
make[2]: *** [../bin/run_lidar2car] Error 1
CMakeFiles/Makefile2:110: recipe for target 'CMakeFiles/run_lidar2car.dir/all' failed
make[1]: *** [CMakeFiles/run_lidar2car.dir/all] Error 2
Makefile:90: recipe for

xiaokn, 2023-07-11:
> 这个splinter-static-3-0是什么模块？要如何安装？每次都报错： /usr/bin/ld: cannot find -lsplinter-static-3-0 collect2: error: ld returned 1 exit status CMakeFiles/run_lidar2car.dir/build.make:391: recipe for target [...]

拉最新的代码，再编译就可以了

zhangyuran-gg, 2024-09-29:
Why did I encounter the following error while executing the cmake steps in a Docker container? I hope you can help me take a look

[image]

zhangyuran-gg, 2024-09-29:
> 为什么在 Docker 容器中执行 cmake 步骤时遇到以下错误？希望你能帮我看一看
> 
> ![1727592735320](https://private-user-images.githubusercontent.com/101116786/371833167-05295bee-2b1d-4a5d-bd84-4e461edc8db3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjc1OTMzOTIsIm5iZiI6MTcyNzU5MzA5MiwicGF0aCI6Ii8xMDExMTY3ODYvMzcxODMzMTY3LTA1Mjk1YmVlLTJiMWQtNGE1ZC1iZDg0LTRlNDYxZWRjOGRiMy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTI5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkyOVQwNjU4MTJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1kMTExYTJhMWZkNDYxNTI0ZTI3NDkwYzJjMzc3N2IyMjYzNDM4MDE2ZTg1MWFmZjNjMzBmMzZlZjdjNDRmMTY0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.TyHxdgY5UbTrd2fARQNKkD3p2ekW7xlhvV8bIKfj8IM)

@xiaokn @yikang-li

zhangyuran-gg, 2024-09-30:
> @xiaokn @yikang-li
Thank you, the issue has been resolved

pedestrain123, 2024-11-28:
@zhangyuran-gg @bodehai请问编译遇到这个错误怎么解决？/usr/bin/ld: /home/wdx/code/perception/sensor_calibarate/SensorsCalibration/SensorX2car/lidar2car/lib/libsplinter-static-3-0.a(bspline.cpp.o): relocation R_X86_64_32S against symbol `_ZTVN8SPLINTER10SerializerE' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /home/wdx/code/perception/sensor_calibarate/SensorsCalibration/SensorX2car/lidar2car/lib/libsplinter-static-3-0.a(bsplinebasis.cpp.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /home/wdx/code/perception/sensor_calibarate/SensorsCalibration/SensorX2car/lidar2car/lib/libsplinter-static-3-0.a(bsplinebasis1d.cpp.o): relocation R_X86_64_32 against `.rodata.str1.8' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /home/wdx/code/perception/sensor_calibarate/SensorsCalibration/SensorX2car/lidar2car/lib/libsplinter-static-3-0.a(bsplinebuilder.cpp.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /home/wdx/code/perception/sensor_calibarate/SensorsCalibration/SensorX2car/lidar2car/lib/libsplinter-static-3-0.a(datatable.cpp.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /home/wdx/code/perception/sensor_calibarate/SensorsCalibration/SensorX2car/lidar2car/lib/libsplinter-static-3-0.a(function.cpp.o): relocation R_X86_64_32S against symbol `_ZTVN8SPLINTER8SaveableE' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /home/wdx/code/perception/sensor_calibarate/SensorsCalibration/SensorX2car/lidar2car/lib/libsplinter-static-3-0.a(mykroneckerproduct.cpp.o): relocation R_X86_64_32 against symbol `__gxx_personality_v0@@CXXABI_1.3' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /home/wdx/code/perception/sensor_calibarate/SensorsCalibration/SensorX2car/lidar2car/lib/libsplinter-static-3-0.a(serializer.cpp.o): relocation R_X86_64_32S against symbol `_ZTVN8SPLINTER10SerializerE' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /home/wdx/code/perception/sensor_calibarate/SensorsCalibration/SensorX2car/lidar2car/lib/libsplinter-static-3-0.a(utilities.cpp.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /home/wdx/code/perception/sensor_calibarate/SensorsCalibration/SensorX2car/lidar2car/lib/libsplinter-static-3-0.a(datapoint.cpp.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: 最后的链结失败: 输出不可表示的节
collect2: error: ld returned 1 exit status