PJLab-ADG/SensorsCalibration #97: unable to download provided sensor2car data

zillur-av, 2023-07-10:
I could not download the data from this link https://pan.baidu.com/s/1r9DcEuvz7IZ_rGpYo_WBBw as it requires some kind of account. Can anybody please provide this lidar to car data?

xiaokn, 2023-07-11:
> I could not download the data from this link https://pan.baidu.com/s/1r9DcEuvz7IZ_rGpYo_WBBw as it requires some kind of account. Can anybody please provide this lidar to car data?
The data for lidar to car is uploaded to Google drive: https://drive.google.com/file/d/1UI19MascaXNNSR1a9SPM9DSC-rmM8IJ7/view?usp=drive_link

zillur-av, 2023-07-11:
Thanks. I see you generated lots of pcd files of one area. Do you have any suggestions on how should I generate my files?