PJLab-ADG/SensorsCalibration #102: Fail to compile for lidar2camera/manual_calib

JiahaoXia, 2023-07-26:
Thanks for your great work!
I follow the instruction on README to download the docker image and then run it.
I want to use the  lidar2camera/manual_calib function.
Here is the CMake error I got:
```bash
CMake Error at CMakeLists.txt:6 (find_package):
  Could not find a configuration file for package "Pangolin" that is
  compatible with requested version "0.8".

  The following configuration files were considered but not accepted:

    /usr/local/lib/cmake/Pangolin/PangolinConfig.cmake, version: 0.6

CMake Error at CMakeLists.txt:10 (find_package):
  Could not find a configuration file for package "PCL" that exactly matches
  requested version "1.9".

  The following configuration files were considered but not accepted:

    /usr/local/pcl-1.9/share/pcl-1.9/PCLConfig.cmake, version: 1.9.1
    /usr/lib/x86_64-linux-gnu/cmake/pcl/PCLConfig.cmake, version: 1.7.2
```
Anyone can help me out of this?

jjunnii, 2023-07-27:
Hi, I have a same isssues. Anyone can help us?

dingkwang, 2023-07-31:
Try to change the required package version to what you have.

Wangmaosen5, 2023-08-05:
请问你解决了吗，我也遇到这样的问题了

ChengBo5, 2023-10-15:
you can change the CMakefile (lidar2camera/manual_calib/CMakeLists.txt) to

`cmake_minimum_required(VERSION 3.5)
project(lidar2camera)

set(CMAKE_BUILD_TYPE "Release")
set(CMAKE_CXX_FLAGS "-std=c++11")
set(CMAKE_CXX_FLAGS_RELEASE "-O3 -Wall -g")
set(CMAKE_CXX_STANDARD 14)

find_package(OpenCV REQUIRED)
include_directories(${OpenCV_INCLUDE_DIRS})

find_package(Pangolin REQUIRED)
include_directories(${Pangolin_INCLUDE_DIRS})

## Get boost
find_package(Boost REQUIRED filesystem system)
include_directories(${BOOST_INCLUDE_DIRS})

find_package(PCL REQUIRED)
include_directories(${PCL_INCLUDE_DIRS})

include_directories(${PROJECT_SOURCE_DIR}/include)

set(CMAKE_ARCHIVE_OUTPUT_DIRECTORY ${PROJECT_SOURCE_DIR}/lib)
set(CMAKE_LIBRARY_OUTPUT_DIRECTORY ${PROJECT_SOURCE_DIR}/lib)
set(CMAKE_RUNTIME_OUTPUT_DIRECTORY ${PROJECT_SOURCE_DIR}/bin)

file(GLOB_RECURSE PARSER_PATH src/*.cpp)
add_library(${PROJECT_NAME} STATIC ${PARSER_PATH})
target_link_libraries(${PROJECT_NAME}  ${PCL_LIBRARIES} ${OpenCV_LIBS} ${Boost_SYSTEM_LIBRARY} ${Pangolin_LIBRARIES})

add_executable(run_lidar2camera src/run_lidar2camera.cpp )
target_link_libraries(run_lidar2camera ${PCL_LIBRARIES} ${OpenCV_LIBS} ${PROJECT_NAME} ${Pangolin_LIBRARIES} libjsoncpp.a)
`
then run cmake .. && make

bilibilicharger, 2024-04-28:
> you can change the CMakefile (lidar2camera/manual_calib/CMakeLists.txt) to
> 
> ```shell
> cmake_minimum_required(VERSION` 3.5)
> project(lidar2camera)
> 
> set(CMAKE_BUILD_TYPE "Release")
> set(CMAKE_CXX_FLAGS "-std=c++11")
> set(CMAKE_CXX_FLAGS_RELEASE "-O3 -Wall -g")
> set(CMAKE_CXX_STANDARD 14)
> 
> find_package(OpenCV REQUIRED)
> include_directories(${OpenCV_INCLUDE_DIRS})
> 
> find_package(Pangolin REQUIRED)
> include_directories(${Pangolin_INCLUDE_DIRS})
> 
> find_package(Boost REQUIRED filesystem system)
> include_directories(${BOOST_INCLUDE_DIRS})
> 
> find_package(PCL REQUIRED)
> include_directories(${PCL_INCLUDE_DIRS})
> 
> include_directories(${PROJECT_SOURCE_DIR}/include)
> 
> set(CMAKE_ARCHIVE_OUTPUT_DIRECTORY ${PROJECT_SOURCE_DIR}/lib)
> set(CMAKE_LIBRARY_OUTPUT_DIRECTORY ${PROJECT_SOURCE_DIR}/lib)
> set(CMAKE_RUNTIME_OUTPUT_DIRECTORY ${PROJECT_SOURCE_DIR}/bin)
> 
> file(GLOB_RECURSE PARSER_PATH src/*.cpp)
> add_library(${PROJECT_NAME} STATIC ${PARSER_PATH})
> target_link_libraries(${PROJECT_NAME}  ${PCL_LIBRARIES} ${OpenCV_LIBS} ${Boost_SYSTEM_LIBRARY} ${Pangolin_LIBRARIES})
> 
> add_executable(run_lidar2camera src/run_lidar2camera.cpp )
> target_link_libraries(run_lidar2camera ${PCL_LIBRARIES} ${OpenCV_LIBS} ${PROJECT_NAME} ${Pangolin_LIBRARIES} libjsoncpp.a)
> ```
> 
> then run cmake .. && make

To correct a mistake, the first line is   cmake_minimum_required(VERSION 3.5)