PJLab-ADG/SensorsCalibration #104: joint_calib

zouzhiwei217, 2023-08-14:
Hello, I made a calibration board to reproduce the joint_calib method. I verified the accuracy of the center detection of the point cloud and took 30 images of the dataset, but the final results were very poor. And I fixed the internal reference of the camera and carried out the experiment, the effect is also very poor. Did you project the point cloud data onto the image, instead of just the four center points?

linClubs, 2023-11-14:
Hi，I have a question. How should this lidar_3d_pts(circle.csv file) be generated？ thanks.

linClubs, 2023-11-14:
sorry,  I got it.

Lpy15540250240, 2023-11-22:
@linClubs hi,bro how to get it ?（circle.csv file） please share it to me. ths bro.

linClubs, 2023-11-25:
@Lpy15540250240 源码自带，哈哈

Lpy15540250240, 2023-11-28:
@linClubs Hi bro， i run it  , just get two pointcloud，not find csv file；Can you give the direct road ?(code linke) hhh ths again

linClubs, 2023-11-28:
@Lpy15540250240  https://github.com/PJLab-ADG/SensorsCalibration/blob/master/lidar2camera/joint_calib/tool/README.md,  
it may not be possible to extract the center point from the PCD file if custom data. Suggest using the data provided by the project for testing

pedestrain123, 2024-11-01:
@linClubs 你好请问您用这个方法测试自己数据集了吗？测试效果怎么样

zw-92, 2024-11-01:
效果非常差

pedestrain123, 2024-11-01:
@zw-92 兄弟，数据包可以发我一份吗？我看看怎么回事，调试一下。方便的话加一下您的QQ,2985782675

pedestrain123, 2024-11-01:
> 时间太久了，数据已经不在了。我建议还是用反光码会更好一点。
> […](#)

好的，那您后面用的是什么方法标定的？

linClubs, 2024-11-06:
> @linClubs 你好请问您用这个方法测试自己数据集了吗？测试效果怎么样
@pedestrain123 手动标定效果不行，因为我雷达是16线的，后面自己用标定板标定了

pedestrain123, 2024-11-06:
> > @linClubs 你好请问您用这个方法测试自己数据集了吗？测试效果怎么样
> > @pedestrain123 手动标定效果不行，因为我雷达是16线的，后面自己用标定板标定了

他的这个joint calib不是也是基于标定板的标定吗？你说的后面用标定板标定不是这个带孔的标定板吗？棋盘格标定板？

linClubs, 2024-11-06:
@pedestrain123 [manual_calib](https://github.com/PJLab-ADG/SensorsCalibration/tree/master/lidar2camera/manual_calib)这个是手动在线标定，你可以试一下

pedestrain123, 2024-11-06:
> @pedestrain123 [manual_calib](https://github.com/PJLab-ADG/SensorsCalibration/tree/master/lidar2camera/manual_calib)这个是手动在线标定，你可以试一下

我正在试，，可以加你QQ后面后面交流一下关于标定和自动驾驶相关吗？QQ：2985782675

pedestrain123, 2024-11-17:
> 你好，我做了一个标定板来复现joint_calib方法，验证了点云中心检测的准确性，拍了30张数据集的图片，但最后的效果很差。而且我固定了相机的内部参考点进行实验，效果也很差。你是把点云数据投影到图片上了吗，而不是只投影四个中心点？

@zw-92 标定板图纸可以分享一下吗？