PJLab-ADG/SensorsCalibration #107: How to get Extrinsic Calibration Matrix from lidar2lidar auto_calib

maxpepi, 2023-08-16:
Hi, 
I want to use your autocalibration function to calibrate multiple lidars, thanks for the great work! I have one question: If I run the auto_calib Program in my docker container it seems to run just fine, I do get the calibration cost and so on in my Terminal. Is there a way to get the Transformation Matrix? Thanks in advance, Max.

maxpepi, 2023-08-28:
If anyone else has the same Problem, I solved it as follows:

as a workaround, i just added

`  std::ofstream outputFile("transformations.txt"); // Öffnet die Datei für den Schreibzugriff 

for (auto iter = refined_extrinsics.begin(); iter != refined_extrinsics.end(); iter++) {
    int32_t slave_id = iter->first;
    Eigen::Matrix4d transform = iter->second;

    outputFile << "Transformation Matrix for slave_id " << slave_id << ":\n" << transform << std::endl;
}

outputFile.close();  `

to the run_lidar2lidar c++ file. Also, you can print the matrix by adding:

` std::cout << "Transformation Matrix for slave_id " << slave_id << ":\n"
              << transform << std::endl; `

maxpepi, 2023-08-28:
For output, see attached Screenshot 
[image]

zhangyuran-gg, 2024-11-07:
> 有关输出，请参阅随附的屏幕截图 ![image](https://private-user-images.githubusercontent.com/119310063/263681582-429fb50b-7cd8-4a11-9db6-fc924a8f0d0a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzA5NTY2OTIsIm5iZiI6MTczMDk1NjM5MiwicGF0aCI6Ii8xMTkzMTAwNjMvMjYzNjgxNTgyLTQyOWZiNTBiLTdjZDgtNGExMS05ZGI2LWZjOTI0YThmMGQwYS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTA3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTEwN1QwNTEzMTJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03ZjQ0ZGNiOWRiNDU3YWIxMzY1NmY0ODU2YWM5YjM5YmVkODZmZDBhYmZlYzY0NjhkNDk0NjI4OTlmMjc3NjY1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.hUZ1JkA69cxDAfZXQ0fBr7mtxS3-PZG2FkLEL_d0SVk)

你好，在你的场景数据中该算法是否准确