PJLab-ADG/SensorsCalibration #108: Pangolin X11: Failed to open X display

POTATO-30, 2023-08-28:
Have you ever encountered this problem？Will you help me？
terminate called after throwing an instance of 'std::runtime_error'
  what():  Pangolin X11: Failed to open X display
Aborted(core dumped)

maxpepi, 2023-08-28:
Yes, try running `x host +`  in a separate Terminal in order to disable the access control.

POTATO-30, 2023-08-28:
> Yes, try running `x host +` in a separate Terminal in order to disable the access control.
Thank you very much for your answer. I tried to run this command on a new terminal and it was prompted that ‘access control disabled,clients can connect from any host', but this issue still exists.

maxpepi, 2023-08-28:
have you checked the points in https://github.com/PJLab-ADG/SensorsCalibration/issues/7 ?

POTATO-30, 2023-08-29:
> have you checked the points in #7 ?
Thank you very much for your answer. I have carefully read this discussion and learned to build an environment step by step on my own host. After a day of hard work, I have finally succeeded! Thank you for your continuous response.

maxpepi, 2023-08-29:
Awesome! Glad it worked out.

AbdullahJirjees, 2024-06-13:
@maxpepi Can you share with me you how did you solved the issue? I tried the xhost + and it still showing the same issue?