PJLab-ADG/SensorsCalibration #115: birdview.hpp  关于left right up down 四点取值问题

lx0928, 2023-09-11:
在birdview.hpp中，关于生成俯视图选取四个像素点与实际设定点的关系
    float left = width_ / 2.8;
    float right = width_ - width_ / 2.8;
    float up = height_ / 2.5;
    float down = height_ - height_ / 25.0;
    bv_pts.push_back(cv::Point2f(left, down));
    bv_pts.push_back(cv::Point2f(left, up));
    bv_pts.push_back(cv::Point2f(right, down));
    bv_pts.push_back(cv::Point2f(right, up));

这个数值是怎么设定的   原始size是1920×1200   如何换成自己的  比如1280×720的话  这些标识点又该如何设定

lx0928, 2023-09-11:
单应性矩阵的对应关系 是图像像素坐标系与哪个坐标系数据相对应来生成单应性矩阵的呢

northeastsquare, 2023-11-23:
2.8, 2.5这个数值可以更改，这里的left， right, up, down指的是世界坐标，把图像平面，映射到这个世界平面上，要求在图片上选点时候，要是平行车道线上的，同时是矩形 。
改变分辨率，不影响。
图像坐标系与地面坐标系的关系。

lx0928, 2023-11-23:
这是来自QQ邮箱的假期自动回复邮件。
 
您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。