PJLab-ADG/SensorsCalibration #116: Strange results "lidar2lidar" automatic calibration

wespenkiller, 2023-09-28:
Hallo.

First of all, thank you very much for this toolbox.

My goal is to calibrate the Lidars, and I wanted to use your automatic algorithm, lidar2lidar. While going through your examples for automatic calibration, I have a question regarding the "initial_extrinsic.txt" in your example. It looks like this:
`device_id: 0`
`(Roll, Pitch, Yaw, tx, ty, tz): 0 0 0 0 0 0`
`device_id: 1`
`(Roll, Pitch, Yaw, tx, ty, tz): 0 0 90 -0.06763169358385032 0.6257701373941718 -0.35145357319239473`
`device_id: 2`
`(Roll, Pitch, Yaw, tx, ty, tz): 0 0 -90 -0.0001307057033816915 -0.4632752877792159 -0.46602840121078765`

Using the provided .pcd files (left.pcd, top.pcd, right.pcd), I assume that automatic calibration should result in calibrated values.
 
I have added this line to the file run_lidar2lidar.cpp to have the transformation results:
`std::cout << "Transformation Matrix for slave_id " << slave_id << ":\n" << transform << std::endl;`

However, the output from the algorithm is as follows:
Transformation Matrix for slave_id 1:

`-0.0235266 -0.995 -0.0970628 -0.00438739`
`0.705173 -0.0853377 0.703881 0.587301`
`-0.708645 -0.0518862 0.703655 -0.397629`
`0 0 0 1`

Transformation Matrix for slave_id 2:

`0.0462854 0.997281 0.0573378 -0.0298915`
`-0.695619 0.0733732 -0.714654 -0.573605`
`-0.716918 -0.00680732 0.697125 -0.426298`
`0 0 0 1`

I am using a Python library to calculate roll, pitch, and yaw angles from the transformation matrices (an example for slave_id 2):

`from scipy.spatial.transform import Rotation as R`
`rotation = R.from_matrix([[0.0462854, 0.997281, 0.0573378], [-0.695619, 0.0733732, -0.714654], [-0.716918, -0.00680732, 0.697125]])`
`roll_pitch_yaw = rotation.as_euler('XYZ', degrees=True)`

So, the angles in degrees are as follows:
For slave_id 1:
`Roll, Pitch, Yaw = -45.00920488, -5.57006802, 91.35449993`
For slave_id 2:
`Roll, Pitch, Yaw = 45.71138557, 3.28703114, -87.34267971`

The results look completely different from the initial_extrinsic values of 0 0 90 and 0 0 -90. When I try to run the algorithm with these calibrated values, the result is some other strange values.

My question is, why does the resulting automatic calibration produce such different values? Perhaps I don't understand something, or the result needs to be interpreted in a certain way?

Thank you for your attention.

Hijiongfeng, 2023-11-17:
"initial_extrinsic.txt" only provides an initial value for calibration. The difference between the final calibration result and the initial result depends on the point cloud. If the difference between the two radar poses is too large, the initial calibration value will be updated when the program runs to calculate the plane normal.

zhangyuran-gg, 2024-11-15:
> “initial_extrinsic.txt” 仅提供校准的初始值。最终校准结果和初始结果之间的差异取决于点云。如果两个雷达位姿之间的差异太大，则当程序运行时将更新初始校准值以计算平面法线。

我使用任何两个点云，即使是两个一模一样的点云进行测试，计算的结果差别很大