PJLab-ADG/SensorsCalibration #121: CMakeFiles/run_lidar2camera.dir/src/run_lidar2camera.cpp.d: No such file or directory

Chiwingum11, 2023-10-24:
cc1plus: fatal error: CMakeFiles/run_lidar2camera.dir/src/run_lidar2camera.cpp.d: No such file or directory
compilation terminated.
CMakeFiles/run_lidar2camera.dir/build.make:75: recipe for target 'CMakeFiles/run_lidar2camera.dir/src/run_lidar2camera.cpp.o' failed
make[2]: *** [CMakeFiles/run_lidar2camera.dir/src/run_lidar2camera.cpp.o] Error 1
CMakeFiles/Makefile2:82: recipe for target 'CMakeFiles/run_lidar2camera.dir/all' failed
make[1]: *** [CMakeFiles/run_lidar2camera.dir/all] Error 2
Makefile:90: recipe for target 'all' failed
make: *** [all] Error 2

Indeed there is no such file but question is, is the cmake .. command responsible for creating the file or not? I have follow the steps and still not able to even get pass the 'make' command line. Please help :)

closed 2023-10-25 by Chiwingum11

geojed, 2023-11-28:
Were you able to solve this?

zayedrmdn, 2024-08-12:
Hi, I've encountered the same problem. Do you know how to solve this?