PJLab-ADG/SensorsCalibration #126: Lidar2camera manual_calib

guoningwang, 2023-11-07:
Hello, thank you for your contribution. I have encountered this problem. Do you know the reason for it?

root@wgn-Precision-7560:/share/lidar2camera/manual_calib# ./bin/run_lidar2camera data/0.png data/0.pcd data/center_camera-intrinsic.json data/top_center_lidar-to-center_camera-extrinsic.json
data/center_camera-intrinsic.json
center_camera-intrinsic
intrinsic:
2109.75 0 949.828
0 2071.72 576.237
0 0 1
dist:
-0.108145 0.138668
top_center_lidar-to-center_camera-extrinsic
Loading data completed!
=>Calibration scale Init!
247
252
width:1920 , height:1200
terminate called after throwing an instance of 'std::runtime_error'
  what():  Pangolin X11: Unable to retrieve framebuffer options
Aborted (core dumped)

AbdullahJirjees, 2024-06-13:
@guoningwang  Hello, I am having the same issue, did you manage to solve it?

lekenzi, 2024-11-20:
@guoningwang Hey, Have you solved the issue yet? I'm having the same issue now.