PJLab-ADG/SensorsCalibration #127: lidar2imu

Chiwingum11, 2023-11-15:
2021-10-26-16-21-29-468 1.000000000 -0.000014703 -0.000006041 0.000061155 0.000014701 1.000000000 0.000004847 0.000095810 0.000006040 -0.000004848 1.000000000 -0.000077579
2021-10-26-16-21-29-568 1.000000000 -0.000031664 -0.000008898 0.000128502 0.000031662 0.999999999 0.000015638 0.000167453 0.000008897 -0.000015638 1.000000000 -0.000122059
2021-10-26-16-21-29-668 0.999999999 -0.000045933 -0.000007882 0.000180359 0.000045932 0.999999998 0.000021805 0.000222766 0.000007880 -0.000021806 1.000000000 -0.000165848
2021-10-26-16-21-29-768 0.999999999 -0.000041261 -0.000027162 0.000239026 0.000041262 0.999999999 0.000018145 0.000246697 0.000027160 -0.000018147 1.000000000 -0.000183308
2021-10-26-16-21-29-868 0.999999998 -0.000042473 -0.000031967 0.000284297 0.000042475 0.999999999 0.000031380 0.000281522 0.000031965 -0.000031382 1.000000000 -0.000197218
2021-10-26-16-21-29-969 0.999999998 -0.000051298 -0.000034643 0.000314884 0.000051298 0.999999999 0.000027465 0.000300088 0.000034642 -0.000027466 1.000000000 -0.000183383

NovAtel-pose-lidar-time.txt

What does each data represent?

DeeKayG, 2024-03-22:
can anyone tell what those 12 values represents and in which order?
My IMU sensor topic echo is in this format:
header:
  stamp:
    sec: 1710008269
    nanosec: 882852351
  frame_id: imu_link
orientation:
  x: -0.006825877990883415
  y: -0.002594882655552313
  z: -0.7126842273063086
  w: -0.7014469802647005
orientation_covariance:
- 0.00042593543332150943
- 0.0
- 0.0
- 0.0
- 0.00038638801831553105
- 0.0
- 0.0
- 0.0
- 0.13248230474005496
angular_velocity:
  x: -0.004857184302894783
  y: 0.008244611230212863
  z: -0.0011483339340475234
angular_velocity_covariance:
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
linear_acceleration:
  x: -0.03084136856920639
  y: 0.03636305442003258
  z: -0.04104681350873415
linear_acceleration_covariance:
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0

program1w, 2024-04-10:
> can anyone tell what those 12 values represents and in which order? My IMU sensor topic echo is in this format: header: stamp: sec: 1710008269 nanosec: 882852351 frame_id: imu_link orientation: x: [...]

我和你的IMU数据一样，你的问题解决了吗，如果你能看到，请＋q 525614957 一起交流一下

DeeKayG, 2024-04-10:
> 
> My IMU data is the same as yours. Has your problem been solved? If you can see it, please +q 525614957 to communicate with us.

I still didn't quite figure out what the heck are those values. I read somewhere in one of these issues where someone mentioned it's R1, R2, R3, T1, R4, R5, R6, T2, R7, R8, R9, T3. I wrote a script to prepare data https://buffalo.box.com/s/9dsmt6b0fp7m6g0qt9fckcc71r38zy97
in this format and still it doesn't work. 
In paper they say our method only takes 19 seconds to calibrate but preparing the data for this package itself takes forever. LOL! So I have changed the package and now I am using ROS1 based LiDAR2IMU Calibration package which works great! 

Here's the proof >> https://buffalo.box.com/s/vmlo1m046tshr5v13jp2v6ibfa6moxfb
I am still refining the parameters and it will get done. It takes 10-15 minutes (WARNING: CPU USAGE AT PEAK) to calibrate depending on how many 8 loops timestamps you are feeding into the algorithm.

The ROS1 bsed package & guide >>  https://autowarefoundation.github.io/autoware-documentation/release-v1.0_beta/how-to-guides/integrating-autoware/creating-vehicle-and-sensor-model/calibrating-sensors/lidar-imu-calibration/

NOTE: While using this ROS1 based package, forget about live visualizing calibration process errors. Let the package do it's job after you execute "roslaunch" command. In the end, you will get calibration results in txt, csv, and PCDs as well.

Good Luck :)

jiheddachraoui, 2025-07-22:
@DeeKayG  hello , your links are expired, can you pls provide new ones?